RobotLocomotion/drake #10551: i always need to run install_prereqs twice on mac

RussTedrake, 2019-01-31:
This has been happening routinely lately.  At least on this newer mac (Mojave, Apple LLVM version 10.0.0)

```
Using dreal/dreal
Using robotlocomotion/director
Installing boost
Installing cmake
Using dreal
Using gcc
Using gflags
Using glew
Installing glib
Using ipopt
Using libyaml
Using lz4
Using nlopt
Installing numpy
Using protobuf
Error: python 3.7.2_1 already installed
Installing python has failed!
Error: python@2 2.7.15_2 already installed
Installing python@2 has failed!
Using scipy
Using tinyxml
Using tinyxml2
Using vtk@8.1
Using yaml-cpp
Installing zeromq
Homebrew Bundle failed! 2 Brewfile dependencies failed to install.
```

Running it a second time basically always succeeds.

jamiesnape, 2019-01-31:
Relates Homebrew/homebrew-bundle#369.

EricCousineau-TRI, 2019-03-21:
Can I ask if this is still an issue?
And have you been able to run things like the diagnostic in the related Homebrew issue, like `brew bundle check --verbose`?

RussTedrake, 2019-03-22:
it has continued, but perhaps something has changed now?  I've updated two macs this morning, and both succeeded on the first try.  happy to close this and can reopen if it happens again?

EricCousineau-TRI, 2019-03-22:
Huh... Well, that works for me!

Will close for now, but we can definitely re-open if it's an issue. Thanks!

RussTedrake, 2019-03-28:
happened again just now.

EricCousineau-TRI, 2019-03-28:
Uh oh!
Can you post of a GitHub Gist of the output, and the output of the diagnostics?

RussTedrake, 2019-03-28:
```
31-33-136:drake russt$ ./setup/mac/install_prereqs.sh
+ source ./setup/mac/binary_distribution/install_prereqs.sh
++ set -euxo pipefail
++ [[ 501 -eq 0 ]]
++ command -v conda
++ command -v /usr/local/bin/brew
++ /usr/local/bin/brew update
Updated 8 taps (bazelbuild/tap, homebrew/cask-versions, homebrew/core, homebrew/cask, homebrew/bundle, homebrew/cask-fonts, caskroom/cask and robotlocomotion/director).
==> New Formulae
buildkit                                             proper                                               protobuf@3.6
==> Updated Formulae
bazelbuild/tap/bazel ✔    clamav                    futhark                   kibana                    opa                       simple-tiles
ffmpeg ✔                  cli53                     gdal                      kubecfg                   orc-tools                 singular
netcdf ✔                  clojurescript             gearboy                   kubernetes-cli            ori                       sip
protobuf ✔                consul                    gearsystem                lf                        osc                       skaffold
python ✔                  coturn                    git-fixup                 libgaiagraphics           osm2pgsql                 spatialite-gui
tbb ✔                     cppad                     github-keygen             libgeotiff                osquery                   spatialite-tools
abcm2ps                   csvkit                    gitlab-runner             liblcf                    oxipng                    spigot
ack                       curl                      glm                       liblwgeom                 parallel                  spirv-cross
acpica                    curl-openssl              glog                      librasterlite             pcl                       stellar-core
aide                      dateutils                 glooctl                   libsoundio                pdal                      step
allegro                   dav1d                     gmt                       libspatialite             pdftoedn                  supersonic
ammonite-repl             derby                     gmt@4                     libssh2                   pgrouting                 swagger-codegen@2
anycable-go               dlib                      gnupg                     libtermkey                phpunit                   swiftformat
apachetop                 doctl                     gnutls                    libtins                   picard-tools              sxiv
arangodb                  doitlive                  gocryptfs                 libtommath                points2grid               termtosvg
armadillo                 dovecot                   googler                   llnode                    polyml                    terragrunt
aubio                     dpkg                      gosu                      logstash                  ponyc                     tokei
auditbeat                 dropbear                  gradle                    lorem                     postgis                   topgrade
autossh                   dssim                     gssh                      mapnik                    presto                    ucloud
awscli                    duply                     hadolint                  mapserver                 procyon-decompiler        vifm
azure-cli                 dynare                    hapi-fhir-cli             maxwell                   proj                      vim
bartycrouch               easyrpg-player            helmfile                  mercurial                 proxychains-ng            vim@7.4
bedtools                  efl                       help2man                  mesa                      ptex                      vtk
berkeley-db               eiffelstudio              homeassistant-cli         metricbeat                pulumi                    vulkan-headers
bettercap                 ejabberd                  hub                       mgba                      pumba                     wsk
bind                      elasticsearch             imagemagick               micropython               pushpin                   wtf
bitwarden-cli             epic5                     influxdb                  mingw-w64                 qxmpp                     xapian
bowtie2                   eslint                    ipython                   mvnvm                     rancher-cli               xcodegen
c-blosc                   exploitdb                 istioctl                  mvtools                   rawtoaces                 xpa
cabextract                ffsend                    jenkins                   nano                      rbenv                     xxhash
caffe                     flatcc                    jid                       nave                      redis@3.2                 ykpers
cdk                       flow                      jpeg-turbo                nco                       redis@4.0                 zabbix
ceres-solver              folly                     json-fortran              ncview                    rke                       zola
cfr-decompiler            fortio                    juju                      nginx                     rst-lint
checkbashisms             freetds                   julius                    nss                       s3ql
chezscheme                frugal                    kallisto                  octave                    sdl2
[... cut ...]
Requirement already satisfied, skipping upgrade: chardet<3.1.0,>=3.0.2 in /usr/local/lib/python3.7/site-packages (from requests>=2.0.0->Sphinx->-r ./setup/mac/source_distribution/requirements.txt (line 2)) (3.0.4)
+ source ./setup/mac/source_distribution/install_prereqs_user_environment.sh
++ set -euo pipefail
++ [[ 501 -eq 0 ]]
++++ dirname ./setup/mac/source_distribution/install_prereqs_user_environment.sh
+++ cd ./setup/mac/source_distribution/../../..
+++ pwd
++ workspace_dir=/Users/russt/drake
++ bazelrc=/Users/russt/drake/gen/environment.bazelrc
+++ dirname /Users/russt/drake/gen/environment.bazelrc
++ mkdir -p /Users/russt/drake/gen
++ cat
```

RussTedrake, 2019-03-28:
on my home machine, too. 
```
...
Installing protobuf
Error: python 3.7.3 already installed
Installing python has failed!
Using python@2
```

RussTedrake, 2019-11-09:
@jamiesnape -- it's possible that it was my underactuated scripts (which you just updated/fixed for me in https://github.com/RussTedrake/underactuated/pull/234) that was fighting the drake setup.  I've only had to run it once on my most recent attempts (hoorah!).  Will reopen this if I ever see the issue again.

closed 2019-11-09 by RussTedrake