RobotLocomotion/drake #10554: Create nightly docker images containing latest binary install and prerequisites

jamiesnape, 2019-01-31:
> Before you post:

> * Have you reviewed the documentation on
[Getting Help](https://drake.mit.edu/getting_help.html)?
> * Have you reviewed existing GitHub issues and StackOverflow posts?
> * Have you rerun `install_prereqs.sh` on your current commit of `drake`?

> If you have encountered a bug (installation, segfaults, etc.), please try to
post reproduction steps and as much relevant output as possible. Consider
posting long snippets as a [Gist](https://gist.github.com/).

EricCousineau-TRI, 2019-01-31:
Xref: [Slack convo](https://drakedevelopers.slack.com/archives/C2PMBJVAN/p1548949781021800)

jwnimmer-tri, 2019-01-31:
... the upshot being "so that new users can take Drake for a spin with as little friction as possible".

jamiesnape, 2019-01-31:
If someone other than me can assign a priority, it would be useful.

EricCousineau-TRI, 2019-01-31:
@RussTedrake Is `priority: med` sufficient, or is this `priority: high` given No. of requests?

RussTedrake, 2019-02-04:
I am going to point people to http://manipulation.csail.mit.edu/install_drake_docker.html for the time being, and see how that works out.  Assuming that goes well, my proposal would be that we have a slightly cleaner version of this available as the first recommendation under our sphinx "installation and quickstart" documentation.

EricCousineau-TRI, 2019-06-13:
Mentioned in #7654.

EricCousineau-TRI, 2019-06-13:
For consumption in `underactuated` via the Dockerfile, Jamie mentioned that he could possibly use the same versioning mechanism (though not sure how it'd work for experiment PRs); though we need to check on restrictions from DockerHub.

EricCousineau-TRI, 2019-09-19:
In service of #11962, targeting this for before end-of-month (Sept. 27), just downloading the binary.

EricCousineau-TRI, 2019-12-05:
Just need to PR docs for Docker Hub: https://hub.docker.com/r/robotlocomotion/drake/tags

PR #12446 (2019-12-05): docker: Add link to published images in documentation
Resolves #10554.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12446)
closed 2019-12-05 by EricCousineau-TRI