RobotLocomotion/drake #10556: Automatically mirror files downloaded by repository rules to S3

jamiesnape, 2019-01-31:
Relates #8595. WIP https://github.com/jamiesnape/drake/tree/mirror-to-s3.

EricCousineau-TRI, 2019-02-07:
Per f2f: Remaining actions are:
* Decide where the scripts are going to live (`drake` vs `drake-ci` vs. something else?).
* Deploy the job. Use Jeremy's magical repository scraping script. Determine job cadence.

tri-ltyyu, 2019-04-11:
Have script running locally to do this. Should be quick issue.

EricCousineau-TRI, 2019-05-16:
Per f2f, delayed due to "firefighting"; will pick up most likely next week.

EricCousineau-TRI, 2019-06-06:
Scripted landed via #11548. Needs to have configuration + jobs land.

jamiesnape, 2019-06-12:
I need to do some refactoring of the CI scripts since the mirroring scripts internally shell out to bazel without the CI-specific flags. Not a huge deal, though.

EricCousineau-TRI, 2019-06-13:
Per f2f, plan is to have a set of particular CI jobs; refactoring mentioned is to put CI-specific Bazel flags to be put as a `bazelrc` file, maybe.

jamiesnape, 2019-07-03:
https://drake-jenkins.csail.mit.edu/job/linux-bionic-gcc-bazel-continuous-python3-mirror-to-s3/

closed 2019-07-03 by jamiesnape