RobotLocomotion/drake #10572: Remove MultibodyTreeContext

sherm1, 2019-02-04:
The internal MultibodyTree initially required a custom Context called MultibodyTreeContext. That is no longer needed and requires dynamic_casts for almost all operations, which can be very expensive. It also makes MBTree unnecessarily complicated.

The only remaining functionality provided by MBTreeContext is knowledge of whether the kinematic state variables are continuous or discrete. That is construction-time information that can be provided directly by MBTree. A few utility methods need to move to MBTree also.

closed 2019-02-05 by sherm1

sherm1, 2019-02-05:
`MultibodyTreeContext`, RIP :ghost:

SeanCurtis-TRI, 2019-02-06:
Alas poor `MultibodyTreeContext`, I knew him well. :skull:

    Hamlet, Act 5 Scene 1

PR #10581 (2019-02-05): Eliminates all traces of MultibodyTreeContext
This gets rid of the last vestiges of MultibodyTreeContext, along with _lots_ of dynamic_casts and other ugly things. The state-access services previously provided by MultibodyTreeContext are now provided by MultibodyTree, and use a just-plain-Context.

This is a follow-on to PR #10569.

Resolves #10572

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10581)