RobotLocomotion/drake #10577: Drake fails when built against the newest libccd

jwnimmer-tri, 2019-02-04:
The error:
```
[4:27:40 PM] ==================== Test output for //examples/manipulation_station:proof_of_life_test:
[4:27:40 PM] No HOME defined in the environment. Will not log.
[4:27:40 PM] terminate called after throwing an instance of 'std::logic_error'
[4:27:40 PM]   what():  Vector appears to be nearly zero.
[4:27:40 PM] ================================================================================
```

jwnimmer-tri, 2019-02-04:
Debugging shows `nhat_BA_W = [-0, -0, -0]` and `depth = 1.474514954580286e-17` for the problematic contact pair (which is for `icontact == 1`).

Run under gdb like so: `gdb --args bazel-bin/examples/manipulation_station/proof_of_life_test --test`.

Probably related to either https://github.com/danfis/libccd/commit/c02958e99344478d80182b46e8894b3d8320a2f2 or https://github.com/danfis/libccd/commit/7931e764a19ef6b21b443376c699bbc9c6d4fba8.

jwnimmer-tri, 2019-02-04:
Over to @SeanCurtis-TRI  for triage (or reassignment).

jwnimmer-tri, 2019-02-04:
I've set this to "Priority: High" in order to understand what the root cause looks like.  In general we want to stay on top of latest masters so we should fix this with some urgency (medium), but knowing _why_ we are failing (and assessing whether _more_ code in Drake it at risk) is the immediate need, in my opinion.

SeanCurtis-TRI, 2019-02-04:
Do you have a sense for why we have the "latest" version of libccd? Or, in other words, what caused our libccd to move?

SeanCurtis-TRI, 2019-02-04:
Nevermind, I typed my question as you were typing the answer. I'm on it.

jwnimmer-tri, 2019-02-04:
All of our CI (public, and TRI-internal) use an older pinned version (from three commits ago).  On the first of every month, I open a Drake PR that updates Drake master to use a newer pin of all of our externals, so that we always stay current.  That _draft_ PR (#10578) is the only place that evidences this failure.

SeanCurtis-TRI, 2019-02-04:
It's just my luck that a dependency that I don't want, which is written in 1983-style C-code *finally* updates in ways that hurts us.  Sheesh....

PR #10740 (2019-02-25): workspace: Bump ccd to latest upstream commit
This includes modifications to `ProximityEngine::ComputePointPairPenetration()`. CCD modified FCL's behavior which left Drake in a bad place. This modification protects Drake from this flaw until FCL is fixed (see fcl issue [#375](https://github.com/flexible-collision-library/fcl/issues/375)).

This replaces PR #10578 and resolves #10577.

For reviewers:
  - The change in proximity_engine.cc is mostly just the inclusion of the tested on line 807 (and the preceding documentation). Everything just got indented.
  - `proximity_engine.h` - a documentat [... cut]
closed 2019-02-25 by SeanCurtis-TRI