RobotLocomotion/drake #10588: Add SparseQuadraticCost

hongkai-dai, 2019-02-05:
As requested by @naveenoid and @amcastro-tri .

Currently if the quadratic cost `0.5 * x' * Q * x + b' * x` has sparse matrix `Q` and `b`, both matrices are stored as dense matrix in `QuadraticCost`. In `GurobiSolver` or `MosekSolver`, we then build the sparse matrix, so Gurobi and Mosek don't do any dense matrix computation.

It would be nicer to provide a `SparseQuadraticCost` that stores the sparse matrix `Q` and `b` directly, without the overhead to convert it to dense matrix in `QuadraticCost`, and convert back to sparse matrix in `GurobiSolver` and `MosekSolver`.

amcastro-tri, 2019-02-05:
Thanks @hongkai-dai. Thus far I don't think this is blocking for us since I might start using Gurobi solvers directly. Also, for our problems our linear constraints are represented by sparse matrices as well.

jwnimmer-tri, 2021-11-11:
This seems like a sub-issue of #4870.  Marking it as such.

jwnimmer-tri, 2024-03-16:
FYI For the linear constraint, we just added `Eigen::Sparse` support directly onto the existing constraint, we didn't make a new one.  That seems like a better approach.