RobotLocomotion/drake #10590: sdf: Improvements to MultibodPlant parser

EricCousineau-TRI, 2019-02-05:
We've started working with OSRC to start to delve a bit deeper into SDF as a format specification. I'd like this to be our public tracker for the overall project to tie separate issues together.

Some things that we are planning to work on:
- Better define edge cases of SDF
- Get better defined semantics for frames `<pose frame='...'/>` ([bitbucket issue](https://bitbucket.org/osrf/sdformat/issues/200)) - successor to #9350
- Enable better encapsulation (beyond gluing stuff with Xacro, or defining a custom format extension)

\cc @sherm1 @scpeters

jwnimmer-tri, 2021-12-15:
@EricCousineau-TRI reading over this issue, I'm not sure about what any remaining call-to-action is supposed to be?  Could you clarify?

EricCousineau-TRI, 2021-12-15:
Remaining action is to polish out composition (use Drake's URDF parsing for composition #15950, then plumbing it through #15948).

Closing this - thanks for the reminder!

closed 2021-12-15 by EricCousineau-TRI