RobotLocomotion/drake #10598: Parser should provide the option of adding to an existing model instance

rcory, 2019-02-06:
Currently, our parser provides `Parser::AddModelFromFile`, which forces the creation of a new model instance on every parse. However, a user may also want the ability to create a single composite model (i.e., contained in a single model instance), by parsing in multiple `sdf` files (e.g., for controlling an arm/gripper composite model using a single inverse dynamics controller).

Having something like `Parser::AddModelFromFileToInstance(file_name, instance_id)` would help serve this purpose.

Related to #10546

soonho-tri, 2019-02-07:
Every issue needs an assignee and I'm assigning it to @amcastro-tri for now. Please feel free to pass it to other people if needed.

amcastro-tri, 2019-02-07:
It'd be nice seeing some example pseudo code for this @rcory. I still don't quite see what problem we are trying to solve. How do you currently work around this? I thought with @edrumwri's recent features you can do this?
With the pseudo code what I am looking for is:
- How do you "add" one model to another? the API above is missing to specify if things weld together, or if there is a joint, on what frames, etc. 
- Could we provide this functionality by "merging" model instances into a single one after we load them with already in master functionality?
- Or would it be just better to get actuation ports and the like for a "group" of model instances?

rcory, 2019-02-07:
> I still don't quite see what problem we are trying to solve. How do you currently work around this? I thought with @edrumwri's recent features you can do this?

In general, I'm reporting this from the point of view of user workflow flexibility. There are ways to achieve the same result (e.g., combining things in a single `sdf` file), but I think it's worth exploring how we can support other workflows that are just as useful.

Let's say I have an arm and a gripper, created by parsing in two separate files (which produces two model instances within MBP). It would be nice to be able to tell MBP that the arm and gripper should be treated as a single model instance (and hence treated as a single model by my controller).

To answer your other questions:
> * How do you "add" one model to another? the API above is missing to specify if things weld together, or if there is a joint, on what frames, etc.

That's a good point. This can probably be debated, but I'd be happy if the model instance was brought in as a floating base (as is already done by `AddModelFromFile`) and have the user add welds/joints/etc. after the fact, if they so choose.

>* Could we provide this functionality by "merging" model instances into a single one after we load them with already in master functionality?

Seems like a reasonable idea.

>* Or would it be just better to get actuation ports and the like for a "group" of model instances?

I suppose one _could_ implement something like this, but it just seems conceptually messier to me than say, having models grouped in the correct model instance to begin with.

jwnimmer-tri, 2022-05-02:
For the record, using SDFormat's recently-added `<include>` feature might be a work-around.

amcastro-tri, 2022-05-03:
@rcory, could you confirm that the SDF `<include>` feature solves your problem?
If so we can close this issue.

rpoyner-tri, 2022-05-25:
I will claim that model composition via SDFormat (include, nesting, etc.), and/or model directives are adequate work-arounds. Having no response from the original plaintiff for nearly a month, closing.

closed 2022-05-25 by rpoyner-tri