RobotLocomotion/drake #10600: Changes to linear_system.cc and linear_quadratic_regulator.cc from #10559 need unit tests

edrumwri, 2019-02-06:
...and a subject area feature reviewer.

jwnimmer-tri, 2019-02-06:
I'll also posit that the correct behavior when linearizing is to skip over disconnected abstract inputs (where Eval returns nullptr), but to revert to the prior behavior of failing when any abstract input was connected in the operating context.

edrumwri, 2019-02-06:
Why not copy the abstract value in the port into the autodiff'd context and add documentation saying that no linearization is conducted over those inputs?

sherm1, 2019-02-06:
Without knowing much about how this is used in practice, I would speculate that it would be more useful to evaluate the abstract input ports and linearize about that nominal point with respect to the _numeric_ inputs. This would be appropriate for example if the abstract input was a "registration port" of some sort and seems (to me) better than refusing to linearize. Disconnecting the abstract port wouldn't work since presumably other nominal calculations would require it.

jwnimmer-tri, 2019-02-06:
For linearize at least, I don't think we can reliably know whether using the `Context<double>`'s abstract values to compute the autodiff dynamics would throw away gradients or not.  In the absence of a sound algorithm to compute the linearization, we should not even try.  (This is why we don't allow for abstract state, also.)

sherm1, 2019-02-06:
@RussTedrake as the author (I think) and Linearize super user (I'm sure) do you have thoughts on what Linearize should do if the target system has abstract input ports?

sherm1, 2019-02-06:
> I don't think we can reliably know whether using the `Context<double>`'s abstract values to compute the autodiff dynamics would throw away gradients or not.

This is a good point, but I think as long as `AutoDiff*double` yields an `AutoDiff` result I think we're OK. For example, in the case Evan is doing we have numeric joint actuation forces u and abstract body forces F. These are combined in the dynamic equations like this: v̇=M⁻¹(u + JᵀF) where M=M(q) and J=J(q). If F is double while everything else is AutoDiff, we can still get a meaningful ∂v̇/∂u as long as the presence of F doesn't prevent linearization.

Are there cases where the presence of a double value like F would prevent differentiation with respect to other variables?

jwnimmer-tri, 2019-02-06:
Hmm, that _seems_ right, but I guess in your shoes I wouldn't want to try to write the proof of why it could never fail, since I believe we don't have any use cases where we are doing this and it's easy to just reject it for now.

However, I think it may be impossible to implement the "Fix the same value" approach you propose right now, anyway?  How do you Fix an abstract input port into `Context<AD>` given only the `InputPort<double>::Eval` value?  It is **not** an invariant of a context that you can copy from AD to double.  For example, in this case if you try to fix `Value<ExternallyAppliedSpatialForce<AD>>` using `Value<ExternallyAppliedSpatialForce<double>>` it will not work.  `Value` does not do scalar casting.

edrumwri, 2019-02-06:
Great discussion- I think unless there is a clear need for this functionality, the "ignore if unconnected" approach is the right one. It appears I'm going to have to make a similar change to enable linearizing MBP with the new applied generalized force input (which isn't always connected) as well. @RussTedrake that means changing the functionality for Linearize so that it ignores unconnected vector input ports, which I think it should do.

sherm1, 2019-02-06:
Yup, an astute observation. `Value<T>` should probably get an update to support `Value<T> = Value<U>` if `T = U` works. But not as part of this project!

RussTedrake, 2019-02-07:
I'm just catching up; sorry I missed the discussion.

The result of Linearize should be a LinearSystem.  If the result is not amenable to the design and analysis used for linear systems, then that's a fail. 

Our implementation of LinearSystem currently only supports (at most) a single input port and (at most) a single output port, it represents those systems with matrices: A, B, xdot0 or xn0, C, D, y0.  I see a few cases:
- Zero input ports: B and D matrices are empty.
- One numeric input port (only): B and D are the linearization of the state/output for this input.
- Multiple numeric input ports: unsupported (but easy to support later).

- If there is an abstract input port, then regardless of if I can autodiff it, I don't know how to write the dynamics as a linear map described by a matrix (representable by matrices B and D).  I don't think it ever makes sense to "linearize" an abstract input.

I understand the proposal to be "treat abstract input ports as though they were constants".  That seems like a slippery slope, but at least it doesn't perform some silently evil work, since the resulting system will not have the same input/output port semantics as the original system.  I can see how it might be useful, as long as it was carefully documented.  To be clear, the only solution that I think makes sense is that the A,B,xdot0,C,D,y0, etc matrices would depend on the value of the abstract inputs.  (Any gradients of the abstract inputs e.g. from autodiff would be irrelevant).

edrumwri, 2019-02-07:
@russtedrake that was my original proposal but @jwnimmer-tri pointed out
that it's not currently feasible. The current proposal is simpler- make
Linearize(.) allow only unconnected abstract-valued inputs.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

RussTedrake, 2019-02-07:
I wrote up the “only unconnected abstract input ports” in my first draft, but then convinced myself that connected vs unconnected was not a major difference.  In both cases, the linear system is created as if the input was pegged (to empty or to a value).

edrumwri, 2019-02-07:
Thanks for verifying that this path is acceptable @RussTedrake

PR #10614 (2019-02-13): Adds an input port for applying generalized forces to MBP. 
Also adds unit tests for Linearize(.) to partially resolve #10600.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10614)
closed 2019-02-13 by edrumwri