RobotLocomotion/drake #10605: dev: Document deprecation policy and tools

EricCousineau-TRI, 2019-02-06:
At present, we have tools for handling deprecations in C++, pybind C++, and Python.
We currently have a very soft documented deprecation policy written, but TTBOMK have a little more stricter policy in practice:
https://drake.mit.edu/code_review_checklist.html?highlight=deprecated#is-the-code-the-minimal-set-of-what-you-want

Ideally, it'd be nice to covert things like:

- Small API changes (deprecations needed?)
- Small + recent API changes (no deprecations needed?)
- Medium changes (deprecations needed?)
- Large changes - e.g. in-place API rework (deprecations needed, changelog?)

jwnimmer-tri, 2019-06-11:
@EricCousineau-TRI per f2f, I'd like to soon have available a function wrapper `PyDeprecated(&my_cxx_method)` to replace `&my_cxx_method` so that the process for deprecating python bindings is a small tweak (adding a decorator call), instead of having to change all of the function pointers into lambdas that call WarnDeprecated.

EricCousineau-TRI, 2019-06-14:
Will try to work on this some tomorrow.

jwnimmer-tri, 2020-05-14:
Per #11012, document should explain what to do about installed data resources (e.g., urdfs).  Current idea is to just mention in release notes and then yank them 3mos later.

EricCousineau-TRI, 2020-07-21:
I think we've documented our tools here:
https://drake.mit.edu/code_review_checklist.html#is-the-code-the-minimal-set-of-what-you-want

We also have the corresponding `WrapDeprecated(&my_cc_function)`, and I've filed the issue #13729 to track that.

Should we close this?

SeanCurtis-TRI, 2020-07-22:
Looking at the "documented tools", I see holes:

1. Deprecating files (e.g., renaming headers, removing headers, etc.)
2. Deprecating XML in URDF/SDF files (we've got old custom tags from the RBT days that are not `drake:` prefixed as well as old behavior that hijacked existing tags but would prefer to use custom `drake:` data.) I haven't spent any time to figure out *how* to do that.
3. Explicit acknowledgment that anything in `internal::` doesn't need deprecation?

jwnimmer-tri, 2020-07-22:
... and even after we plug those holes, we still should have a playbook that walks authors (and reviewers) through it step by step with examples, which does not exist yet.

SeanCurtis-TRI, 2020-07-22:
One thousand times, "YES!"

jwnimmer-tri, 2020-11-18:
We should capture a known problematic case: perfect forwarding defeats the compiler annotations, as in `builder.AddSystem<FooSystem>(deprecated_args)`.

rpoyner-tri, 2021-01-25:
A possibly interesting example for the future playbook is #14538 .

EricCousineau-TRI, 2021-01-25:
See also: https://github.com/RobotLocomotion/drake/pull/14546

mitiguy, 2021-02-09:
Although the goal of PR #14538 was a simple name change, the deprecation process was not obvious to me.  I created [notes ](https://docs.google.com/document/d/1n5CK3c8TLn6aqHgT4XnI_oZBu2AwRxTRWgJjOAKhYXE/edit?usp=sharing ) that may help others and/or be modified to be part of a deprecation playbook.

@jwnimmer-tri  Please let me know if you think it helpful to create a PR to add this document to part of the Drake website.

jwnimmer-tri, 2021-02-09:
> ... I wondered if you think it helpful to create a PR to add this document to part of the Drake website ...

In general, I do think that having a clearer playbook would benefit us (as I noted above in https://github.com/RobotLocomotion/drake/issues/10605#issuecomment-662456506).

However, I'm having trouble parsing out exactly what you're proposing based on the message above.  If the implication is that @mitiguy would create such a PR -- then, no, I don't think that would be useful.

mitiguy, 2021-02-11:
OK -- No PR from me.  I 'll leave a link to the updated [CppPythonDeprecateAPI document here](https://docs.google.com/document/d/1n5CK3c8TLn6aqHgT4XnI_oZBu2AwRxTRWgJjOAKhYXE/edit?pli=1).  Hopefully, it informs  this issue of text that would have been benefited me in performing a simple API name change.  In hindsight, this simple name change took longer than I expected because I did not have the proper mix of speciality code.

jwnimmer-tri, 2021-11-11:
When we next return to this issue, we need to split it in twain:
- policy documentation for stable releases
- developer tooling to making deprecating things easier

They are vastly different topics.

jwnimmer-tri, 2021-12-01:
Most of this issue revolves around the supporting tooling and how-to documentation, so I'll use this issue to track that aspect.  I'll open a new issue => #16159 to track the yes/no of the policy.

jwnimmer-tri, 2022-03-14:
FYI on a nice trick worth putting into our playbook:

When deprecating a helper struct:

(1) Change the struct name from `MyFoo` to `DeprecatedMyFoo`.

(2) Alias `using MyFoo DRAKE_DEPRECATED(...) = DeprecatedMyFoo;` so that users will get a notice.

(3) Internal to drake, spell the struct as `DeprecatedMyFoo` directly, to avoid first-party warnings.  Directly deprecate any public functions that mention it.

jwnimmer-tri, 2022-03-14:
Another tip:

When replacing a large API surface (and deprecating the old API surface), the easiest way to preserve the old-API tests while also duplicating and modifying them for the new-API is copy each unit test into a duplicate file:

https://github.com/RobotLocomotion/drake/pull/16739#pullrequestreview-908233511

jwnimmer-tri, 2023-05-23:
Another tip: when adding a new field to a struct, give it a default value:
https://reviewable.io/reviews/RobotLocomotion/drake/19145#-NSfYQ168nEKK_7dn3jS