RobotLocomotion/drake #10606: ci: Determine timeline for Xenial support

EricCousineau-TRI, 2019-02-06:
Should determine when to remove support.

\cc @jamiesnape

jamiesnape, 2019-02-07:
I will propose either October 26th, 2019 or April 26th, 2020 as an arbitrary date to switch off the last build with the turndown being three to six months before the chosen date.

EricCousineau-TRI, 2019-02-07:
Also need a date for Python 2. Ideally, Jan. 1, 2020, since that's its EOL. (TODO: Add authoritative xref.)

jwnimmer-tri, 2019-02-07:
I think dropping Xenial toward the end of 2019 is a good goal (Oct 26th is fine).  That will make room for working toward 20.04 LTS support after the new year.

EricCousineau-TRI, 2019-02-14:
Per f2f, should define a date to make `python2` the explicitly needed call-out in build jobs, and make `python3` the default. For now, on or before June 15, 2019.

Should also consider when to make platforms default to using Python 3. Could happen before or at the same time as build job switch. Min. 3 months before removing Python 2 support.

jwnimmer-tri, 2019-03-01:
@EricCousineau-TRI notable for the TRI side, we should endeavor to move past GCC-6 on Bionic internally, prior to GCC-5 (Xenial) disappearing from Drake's CI.  In Drake CI only has GCC-7, we don't want to be firefighting GCC-6 issues in our internal CI.  GCC-7 had major C++17 improvements, so GCC-6 and GCC-7 are relatively dissimilar.

EricCousineau-TRI, 2019-03-07:
Notes from f2f Kitware meeting:

Might also be able to push up Python 2 deprecation on Mac.
Would simplify Homebrew packaging, and lighten the load on Jamie for packaging.

Possibly just make Mojave images be Python 3 only?
Would relate Drake Visualizer upgrade, but that may not be too much overhead - all platforms should use Drake Visualizer on Python 3.

EricCousineau-TRI, 2019-06-20:
Question from @jamiesnape:

1. Where supported, can we default to Python 3?
2. Can we drop Python 2 the same time as Xenial?

My personal vote:
(1) - yes
(2) - yes, depending on @RussTedrake's needs for `underactuated` / `manipulation`?

RussTedrake, 2019-06-21:
My vote:
(1) yes.
(2) yes.

RussTedrake, 2019-06-21:
@peteflorence asked "Does anybody know what the ROS’s Python 2 —>3 plan is?"  Could we spell out any implications for ROS users here?

jamiesnape, 2019-06-26:
You can get it to (mostly) work if you build from source, otherwise time to migrate to ROS 2.

EricCousineau-TRI, 2019-06-26:
Yup, that's what we do in Anzu for things that require C extensions like `cv_bridge` and `tf2_py`. Pure 
Python packages just work out of the box in Python 3 (although you have Python 2 libraries on your path...).

Regarding Python 2 vs 3 defaulting in [Slack discussion](https://drakedevelopers.slack.com/archives/C3YB3EV5W/p1561556826021000):

I would say we initially try to target July 15. I will add a checkbox above.

EricCousineau-TRI, 2019-06-26:
Given votes and per discussion in #11729, updated timeline to push up Python 2 removal to the same date as Xenial.

EricCousineau-TRI, 2019-06-27:
When we drop Python 2, we should use Director's Python 3 (changing branch, etc.).

jamiesnape, 2019-07-15:
CI will be switching defaults either later today or tomorrow. I just need to create Mac `-python2-` jobs.

EricCousineau-TRI, 2019-07-18:
Jamie mentioned that CI is now switched over to Python 3 default. Just need to fixup merge in #11733.

EricCousineau-TRI, 2019-07-22:
Just saw this ROS discourse post recently, but to @peteflorence's question about Python 2 + ROS1:
https://discourse.ros.org/t/ros-1-and-python-2-7-end-of-life/9941/6
Takeaway is that Python 3 support for ROS 1 won't really come until Noetic, which will have alpha releases around October, but seems like it'll be coupled to Ubuntu 20.04 and is targeted to release in May 2020?

I haven't yet figured out what that means for Drake; my ideal minimum-effort solution is that we still only accommodate Python 3 after the October deadline, meaning that it would cut off use of ROS1 prebuilt binaries with Drake.

However, from my experience in Anzu with our limited use of ROS1, it seems like most core pure Python ROS1 packages support Python 3, and most C extension libraries (e.g. `tf2_py` and `cv_bridge`) can be rebuilt in Python 3. These can be used as an overlay on an existing Python 2 installation, so the only finicky part is ensuring you map the upstream Python 2 dependencies to Python 3. (I will see if I can post the Docker recipe we have for producing the Python 3 C extension overlay in that ROS discourse post.)

On the ROS 2 front, it seems to be up and coming, but the translation is still not completely trivial, and there are some remaining RViz features to be ported that had stopped me during my spike test. There is some active work on this, and I will revisit it later either this quarter or next.

---

In short: It's, uh, complicated, given ROS1's scheduling having dead-time to support Python 3 after Python 2's EOL, and possibly not supporting 16.04 / 18.04 for Noetic... :(

jamiesnape, 2019-10-01:
Note, I need to wait for a release to switch off most of Mac Python 2 support.

jwnimmer-tri, 2019-10-01:
Ah, I didn't notice that Oct 1 was a drop date for that support.  I'll aim for a release in a next couple days, then.

jwnimmer-tri, 2019-10-02:
The v0.11.0 release is pushed; I think we can start turning down Python 2 on macOS now.

jamiesnape, 2019-10-17:
Python 3.8 is out. We assume 3.7 for Mac in various places, so that will need fixing ASAP. More fun (for me).

jamiesnape, 2019-10-17:
Watching https://github.com/Homebrew/homebrew-core/pull/45337.

EricCousineau-TRI, 2019-10-31:
Given that we have tracking issues for the remaining two items (Catalina + Director), closing this given that we've generally dropped Python 2 + Xenial.

closed 2019-10-31 by EricCousineau-TRI