RobotLocomotion/drake #10608: python: Include bindings in valgrind anaylsis

EricCousineau-TRI, 2019-02-06:
Example using re-compiled Python:
https://gist.github.com/EricCousineau-TRI/ce79d3265bb72934267e24ddc8c623bc

This would have to use `pip` or something else to "recreate" the packages, which would be painful.

(Moving from Kitware notes to GitHub issue.)

EricCousineau-TRI, 2020-05-02:
Was this already done via the `helgrind` mods?

jamiesnape, 2020-05-04:
Only partially (not for memcheck: https://github.com/RobotLocomotion/drake/blob/7a03e312cea8bfb1ea2177c1ab49312d577d9c7a/tools/dynamic_analysis/bazel.rc#L215).

I can't imagine us ever recompiling python, so the task is more finishing the suppressions to be able to use `python3-dbg`.

jwnimmer-tri, 2021-11-11:
Closing as "nice to have, but unlikely to ever float to the top of the priority list".

closed 2021-11-11 by jwnimmer-tri