RobotLocomotion/drake #10611: cdash: Recover dynamic analysis summaries

EricCousineau-TRI, 2019-02-06:
(From Kitware notes.)

Dynamic analysis summaries disappeared from CMake to Bazel, could get them back in. (Info is there, just need to parse.)

jamiesnape, 2019-02-07:
Switching to "low priority" since it is technically in progress.

BetsyMcPhail, 2019-02-07:
[CDash issue](https://github.com/Kitware/CDash/issues/795)

tri-ltyyu, 2019-04-11:
This can be backlogged because the impact isn't felt right now. Please comment if you're affected.

jwnimmer-tri, 2020-05-14:
Could I see an example of these summaries (maybe for a different project)?  Is it like statement coverage?  The dynamic analysis facts we care about are hard-fail tests in both pre-merge and post-merge, so I don't immediately see what beyond that we'd want?

jwnimmer-tri, 2020-12-09:
Given that there doesn't seem to be anything still missing (kcov works well), I'll close this sufficiently done.

closed 2020-12-09 by jwnimmer-tri