RobotLocomotion/drake #10615: multibody doxygen gives people the impression that MBP does not support collision

RussTedrake, 2019-02-07:
I've had a few people look at this page:
https://drake.mit.edu/doxygen_cxx/group__multibody.html

and say "so MBP doesn't support collision yet?" (because the only section on collision says "RBP only").  That's not the message we want to send!

sherm1, 2019-02-07:
@SeanCurtis-TRI I think this doc is yours -- time for an update?

SeanCurtis-TRI, 2019-02-07:
I'm on it.

SeanCurtis-TRI, 2019-02-07:
I'd recommend splitting this:

- the "Collision Filter Concepts" belong to me.
- the "Compliant Contact in Drake" belongs to @amcastro-tri (as the time-stepping contact in MBP is no longer what is documented in the doxygen.)

sherm1, 2019-02-07:
The immediate need is to avoid the misleading impression that MBP doesn't support contact. A larger reworking could be deferred if necessary.

SeanCurtis-TRI, 2019-02-07:
The problem is that much that is there is not correct for MBP/SG. So, the simplest fix is to delete the module entirely and then fill it in after the fact.

All in favor?

SeanCurtis-TRI, 2019-02-07:
This would also resolve #9467.

sherm1, 2019-02-07:
Aye! (Should still say _something_ about MBP and contact -- maybe just a link to some Doxygen or example?)

PR #10622 (2019-02-11): Removed RBP-related contact info
1. Explanation of two forms of contact filtering is not necessary for SG. So, the contrast of these two approaches has no relevance going forward.
2. The contact model in MBP is not the same as is documented here.

Rather than sequestering these somewhere to facilitate future use of RBP, we might as well just remove them -- git remembers.

resolves #9467
resolves #10615

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10622)
closed 2019-02-11 by SeanCurtis-TRI