RobotLocomotion/drake #10616: Symbolic expressions from random distributions with the same parameters are not equal

sammy-tri, 2019-02-07:
Expressions containing random variables from distributions with the same parameters are not equal according to `ExprEqual` even though their string values are identical.

Test code:
```
    normal_distribution<Expression> d1(1.0, 2.0);
    normal_distribution<Expression> d2(1.0, 2.0);
    EXPECT_PRED2(ExprEqual, d1(), d2());
```
results in:
```
ExprEqual(d1(), d2()) evaluates to false, where
d1() evaluates to (1 + 2 * random_gaussian_0)
d2() evaluates to (1 + 2 * random_gaussian_0)
```

soonho-tri, 2019-02-07:
This is an expected behavior. You have two distinct `normal_distribution`, `d1` and `d2`.  As a result, the expressions generated by them are not identical.

> even though their string values are identical.

FYI, symbolic variables are identified by their IDs, not their names. I can change the implementation to generate less-confusing names such as:
```
d1() evaluates to (1 + 2 * random_gaussian_0_0)
d2() evaluates to (1 + 2 * random_gaussian_1_0)
``` 
by including the source ID (distribution).

sammy-tri, 2019-02-07:
Ah, ok.   My original use case was writing a unit test to make sure that a function which returned a `symbolic::Expression` had formulated the expression correctly.  Is there a recommended way to do this (which isn't using `ExprEqual`)?

soonho-tri, 2019-02-07:
I hope the unit tests that I wrote in https://github.com/RobotLocomotion/drake/blob/master/common/test/symbolic_expression_test.cc#L2261-L2420 can help. If not, could you share your WIP commits with me so that I can learn the context and give you a better advice?

sammy-tri, 2019-02-07:
Thanks, I think I can make that work for my needs.  Printing the source ID might be helpful in the future to avoid confusion, but I don't think it's necessary now unless you're particularly inclined to make the change.  Feel free to close the issue as working as intended if you like.

soonho-tri, 2019-02-07:
OK. Let me close this for now. If a similar issue comes out, I'll improve it then.

closed 2019-02-07 by soonho-tri