RobotLocomotion/drake #10619: ci: leak-san required builds can be slow (~2hrs)

EricCousineau-TRI, 2019-02-07:
See Slack convo: https://drakedevelopers.slack.com/archives/C270MN28G/p1549493207082300

Difference between failed download and a very slow download.
May be resolved once this makes it into a release (and we use it):
https://github.com/bazelbuild/bazel/issues/5440

Possible solutions:
* Switch off remote caching for leak-san
* Something else?

jwnimmer-tri, 2021-01-25:
Now that S3 guarantees strong consistency, using S3 directly for our remote cache is probably going to be more performant than our current EC2 solution.

We should investigate that as a solution.  We could even roll it out for Ubuntu builds only, in case its untenable to immediately switch the macOS build farm to using it.

jwnimmer-tri, 2022-01-27:
Another option: split the cache into even more, different servers (e.g., a dedicated LSan-only one).

jwnimmer-tri, 2023-01-28:
I'm not sure if there is still more to do here.  The builds have seemed pretty healthy to me lately, I haven't seen any pre-merge stalls on any PR's I've been tracking, since many months ago.

Maybe @svenevs can keep an eye on this as we update our caching infra, and then close it out.

closed 2023-02-16 by jwnimmer-tri