RobotLocomotion/drake #10624: doc python: Add instructions for debugging bindings from CMake install

EricCousineau-TRI, 2019-02-07:
Per f2f with @edrumwri:
For debugging bindings, binding under Bazel is sufficient per the [current dev docs](https://drake.mit.edu/doxygen_cxx/group__python__bindings.html).

However, external users are likely to run into mysterious segfaults, or just odd behavior in which they may want to step through the code.

The action item here is to add a minor section in the [user python docs](https://drake.mit.edu/python_bindings.html) for how to debug with bindings built from CMake. Ideally, it should show how to get the SHA1 from a downloaded binary for traceability.

edrumwri, 2019-02-07:
To elaborate on one aspect:

The workflow that I am imagining for typical pydrake users is downloading the binaries, setting the PYTHONPATH, and running their python script a la `python myscript.py`.

We currently lack documentation for how to debug python scripts into the C++ code, which is currently often necessary, _without moving myscript.py into a Drake subdirectory and creating a bazel build file for it_. I do not believe users will enjoy having to do this.

jwnimmer-tri, 2022-05-06:
@EricCousineau-TRI what do you imagine is the solution path we need to document here?  As best I understand, you're proposing to teach something like this:

(1) Find the version of the precompiled binary they were already using.

(2) Check out drake.git from that tag (or sha).

(3) Build & install pydrake from source, with debugging and/or symbols enabled.

(4) Use the source-installed copy, instead of the binary, when running their own project.

EricCousineau-TRI, 2022-05-08:
Yup! Perhaps with the added Step 0 to address Evan's point:

(0) Show running the script under `gdb` / `lldb` using (non-symbol) binary installed Drake, and get a rough stacktrace. (alternatively, using `coredumpctl gdb` to get the trace post-portem)

jwnimmer-tri, 2022-05-09:
So just `gdb --args python3 myprogram.py`?

jwnimmer-tri, 2024-03-19:
My prior comment shows the simple command that works just fine. Nothing here is Drake-specific.

Closing as not planned.

closed 2024-03-19 by jwnimmer-tri