RobotLocomotion/drake #10625: pydrake OutputPort.Eval should return np.ndarray (not BasicVector)

jwnimmer-tri, 2019-02-07:
To better align with all of the forthcoming simplifications of systems framework evaluation, and because it's more useful in the first place, `pydrake.systems.OutputPort.Eval` should return `np.ndarray`, not `BasicVector`.  As it stands now, almost all uses just call `.get_value()` (to obtain the ndarray) immediately after the Eval.

This is breaking change, and we are not planning on offering any deprecation period or warning shim.  To get the old behavior call `my_port.EvalBasicVector` instead (#10623).

jwnimmer-tri, 2019-02-07:
\CC @RussTedrake FYI this might affect some of your downstream pydrake uses.  I'd say the ETA for the change is 2019-02-13 (i.e., early/mid next week).

RussTedrake, 2019-02-07:
I'll make the fixes when it happens.

jwnimmer-tri, 2019-02-12:
The ETA for this to land is mid-/late- day tomorrow (Tue 2019-02-12).

jwnimmer-tri, 2019-02-12:
Fixed in #10592.

jwnimmer-tri, 2019-02-12:
For pydrake users that want to make similar updates, and good example is [this diff](https://github.com/RobotLocomotion/drake/commit/57a301889960b8d5fce4b9d2d838f2e12a763d43#diff-8d0e43adb57170eddda336a62e5ac996) to joint_teleop.py.

closed 2019-02-12 by jwnimmer-tri