RobotLocomotion/drake #10631: Upgrade to latest Doxygen

jwnimmer-tri, 2019-02-08:
The latest Doxygen (1.8.15) has a few bugfixes that end up being very nice (in particular, using declarations for class methods).

Homebrew already has 1.8.15, and in general seems like it will track the latest releases like it usually does.  Ubuntu is stuck on 1.8.13 (even in > 18.04).  We should provide a binary image for latest on Ubuntu, so that we can have nice things.

jwnimmer-tri, 2019-02-08:
I can't remember if SourceForge still adds malware to its downloads, but they do have builds up at https://sourceforge.net/projects/doxygen/files/rel-1.8.15/ that we could mirror for ourselves.  Upstream also provides a binary release, but it's unsigned and _very_ slow to download.  We could also build our own release tarball or deb from source.

jwnimmer-tri, 2019-02-08:
@jamiesnape Do you have any suggestions on how we should obtain (and refresh) a binary release of Doxygen for Ubuntu?  Just running CMake seems to work fine for me -- we could write a short docker script to make a binary, maybe?  Or use one of the prebuilts mentioned above.  Or...?

\CC @EricCousineau-TRI

jamiesnape, 2019-02-08:
Does http://doxygen.nl/files/doxygen-1.8.15.linux.bin.tar.gz work on 18.04? If not, I guess I can build binaries and host them in S3 like VTK.

jwnimmer-tri, 2019-02-08:
I will try to check today.  When I tried to fetch it last night, the download was going to take hours.

jamiesnape, 2019-02-08:
I will start a download here and upload to S3, but I guess I could probably build it quicker than the download takes.

jamiesnape, 2019-02-08:
Do we need clang-assisted parsing? The Homebrew versions don't support it by default.

jwnimmer-tri, 2019-02-08:
Nope.  It's currently set to OFF in our Doxyfile, and in my local testing it seems fine to keep it that way.

jamiesnape, 2019-02-08:
Try these:

* https://drake-packages.csail.mit.edu/doxygen/doxygen-1.8.15-bionic-x86_64.tar.gz
* https://drake-packages.csail.mit.edu/doxygen/doxygen-1.8.15-xenial-x86_64.tar.gz

I will PR the build scripts if/when we use them.

jamiesnape, 2019-02-08:
I probably should add a license file to archive, so the SHA's may change when I fix that.

PR #10659 (2019-02-12): workspace: Use newer Doxygen on Ubuntu
Closes #10631.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10659)
closed 2019-02-12 by jwnimmer-tri