RobotLocomotion/drake #10636: `MBP::GetMutableFoo()` methods can dangerously bypass cache invalidation.

amcastro-tri, 2019-02-08:
For instance, if I do `auto my_qs = MBP::GetMutablePositions(context)` to retrieve a reference to the positions vector, everything depending on q's will get invalidated. Say I compute a pose and therefore position kinematics gets validated. If I later do `my_qs.setZero()`, nothing depending on qs will get invalidated leading to very obscure and difficult to track faulty computations. 

The proposed solution is to deprecate the `GetMutableFoo()` methods in favor of `SetFoo()` methods.

amcastro-tri, 2019-02-08:
@edrumwri, you worked on these in the past, what do you think?

jwnimmer-tri, 2019-02-08:
The method already has [pretty scary doxygen](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#ace2984079483027345739f5e1bc8a19c).  Did you happen to step on a landmine anyway, and that's why this is coming up now?

amcastro-tri, 2019-02-08:
True, thanks for pointing that out @jwnimmer-tri. Yes, I did step on the landmine of finding this method before its sibling (precisely and oddly when adding more MBP cache entries). 
Why do we even have this method in there if its so dangerous then? why not just only provide `SetFoo()`?

jwnimmer-tri, 2019-02-08:
Well, at least right now its useful because `Set` forces an allocation or copy (because it's argument is a a `const Eigen::Vector&` not an `Eigen::Ref`).  Even if that's repaired, I could imagine Mutable still being useful if the user wants to invalidate the entire position state at once, and then populate it model-by-model or body-by-body piecewise (possibly for just a few bodies) -- so you're not copying the whole q every time you just want to change a single pose.

jwnimmer-tri, 2019-02-08:
It's possible that in addition to the scary doxygen, we should make the method names even scarier also.  But this question applies to _all_ of the GetMutable methods such as the Context, State, etc.  I thought that @sherm1 already had an issue about that, but I'm not immediately seeing it.  We should probably take a consistent solution throughout.

sherm1, 2019-02-08:
> Why do we even have this method in there if its so dangerous then?

These are primarily useful for writing integrators where making state copies every substep is a bad idea. In my upcoming PR for cache-friendly integrators you can see how they are supposed to be used. They are super dangerous though and should all be marked advanced (if they aren't already) and get CamelCaseNames rather than the friendly_names_they_have_now. (#9205 has that has a checkbox item.)

sherm1, 2019-02-08:
Oh, actually I thought you were talking about the similar methods on Context. Now I see these are MultibodyPlant methods and I do wonder why we should have them at all.

jwnimmer-tri, 2021-11-12:
I did a quick grep.  Every example I found that was using these wasn't even storing it in a named temporary, it was immediately writing a new value.  The `SetPositions()` already all take an `Eigen::Ref`.  So, it seems like we should just deprecate these and move on.

xuchenhan-tri, 2023-07-12:
I can clean this up in the coming weeks unless there's objection here for deprecating these functions.

jwnimmer-tri, 2023-07-12:
My suggestion would be to grep around and re-confirm my premise from two comments prior (that all existing call sites would be equally happy with `SetFoo`, because none of the retain the return value beyond the C++ statement anyway).  If that still holds, to me it would be fine to deprecate for removal.

PR #20424 (2023-11-01): Deprecate get mutable state functions in MultibodyPlant
Closes #10636

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20424)
closed 2023-11-01 by SeanCurtis-TRI