RobotLocomotion/drake #10639: Inverse dynamics profiling reveals cost of dynamic_cast

amcastro-tri, 2019-02-08:
Recent profiling of Inverse Dynamics reveals that over 9% of the cost is lost with `dynamic_cast`. 
Many of these are spread throughout the `system::` framework. 
Inverse dynamics is a fundamental component of our multibody engine and is used in many other ways than just of ID calculations. Currently for instance, we use it to also compute the mass matrix.

The proposal is to switch to `static_cast` whenever it's safe within all of drake. That could be accomplished with a `drake::safe_dynamic_cast` that resolves to a cheap `static_cast` in release builds.

I assigned myself this only with the objective of finding someone more capable than myself to do this. Volunteers?

sherm1, 2019-02-08:
I would prefer to call this something like `fast_dynamic_cast`. The original `dynamic_cast` is the safe one -- we're looking for one that is _less_ safe.

BTW note that this _is_ still a dynamic cast -- in Debug it will only work where dynamic_cast is allowed. In Release we just give up the runtime checking.

sherm1, 2019-02-08:
The easiest implementation is a macro:
```c++
#ifdef DRAKE_ASSERT_IS_ARMED
  #define DRAKE_FAST_DYNAMIC_CAST dynamic_cast
#else
  #define DRAKE_FAST_DYNAMIC_CAST static_cast
#endif
```
Does anyone object to that or have a better idea?

jwnimmer-tri, 2019-02-08:
From similar ABSL functions https://github.com/abseil/abseil-cpp/blob/master/absl/base/casts.h, it doesn't look _that_ hard to write it as a function, does it?  We could imagine improving the error reporting in the debug case.  Are you saying you've done it before and the template function ends up being difficult?

sherm1, 2019-02-08:
Nope, I've only done it as a macro. A template function would be prettier, though I'm not sure it would be better. The Abseil ones are actually new casts so do useful compile-time checks on the operation -- we would likely still have to depend on dynamic_cast to do the checking for us.

jwnimmer-tri, 2019-02-08:
Well, a function would let us call it `fast_dynamic_cast<T&>` at least, instead of screaming all the time.

jwnimmer-tri, 2019-02-08:
I seem to recall reading recently too about a faster dynamic-ish cast.  Something like that because C++ dynamic_cast has to do lots of work to account for possible type lattices, but if you are just going from MyBase to MyFinal then you can do faster things that still fail fast(er) than static_cast.  I don't remember if that was just `DRAKE_DEMAND(&typeid(from) == &type(MyFinal))` + static_cast, or what.  I'll see if I can find the reference.

sherm1, 2019-02-08:
I'm on the fence about whether _not_ screaming about an unchecked dynamic cast is actually better.

jwnimmer-tri, 2019-02-08:
What about `debug_checked_static_cast`?  It's basically a static_cast + assert, so we shouldn't be calling it `dynamic_cast` anyway.

sherm1, 2019-02-08:
I don't like that because this is really a dynamic cast (only applicable to polymorphic classes). static_cast can be used in more circumstances and most of those _cannot_ be checked by dynamic_cast.

jwnimmer-tri, 2019-02-08:
`unchecked_dynamic_cast`?

sherm1, 2019-02-09:
I like `unchecked_dynamic_cast`! Still not sure it shouldn't scream DRAKE_UNCHECKED_DYNAMIC_CAST but I don't think I can resist the elegance of the template function.

sherm1, 2019-02-09:
Does anyone want to volunteer to write `unchecked_dynamic_cast<T>()` ?

jwnimmer-tri, 2019-02-09:
> Recent profiling of Inverse Dynamics reveals that over 9% of the cost is lost with `dynamic_cast`.
> Many of these are spread throughout the `system::` framework.

@amcastro-tri Can you cite one or a few specific examples of the casts that are slow?  If we can keep the typechecking but make it fast, it seems nice to have good error messages, even in release mode.

jwnimmer-tri, 2019-02-09:
(See, for example, #10649.)

amcastro-tri, 2019-02-11:
> Can you cite one or a few specific examples of the casts that are slow?

I'd need to run the profiling again @jwnimmer-tri, what I reported were results including all dynamic casting in Drake. #10644 seems to reveal a particular one. I'll investigate more.

jwnimmer-tri, 2019-03-04:
Having had more time to think about this, I am not super keen on the idea of `unchecked_dynamic_cast` anymore.  We might eventually want it, but I would like to defer as long as possible.

I think two cases will cover us for _most_ uses of `dynamic_cast`:

(1) In certain methods, we can change to `static_cast`, even in Debug builds.  For example, if there is a private class invariant where we _know_ through direct, local reasoning that the downcast will always succeed.  The user cannot make any API calls that violate the invariant, so we don't need to explicitly check for "errors".  Our CI memory checkers / ubsan / etc are good enough.

(2) In cases like `Value<T>` where the user can store a T-value and then later retrieve it by mentioning `T` again to induce a downcast, we _must always_ check that the user supplied T correctly.  Almost no users (including our developers) run with assertions enabled.  We _must_ have a diagnostic always turned on, in cases where the user can make a mistake.  We just have to make that diagnostic as fast as necessary / practical.  I have done this for `Value<T>`.  I am not sure that there are any additional uses where even a naively-implemented check would have a speed impact that matters.

Basically, having read more about this, I now subscribe the C++ school that "There are no valid uses of dynamic_cast" -- that all of them are covered by either the above two items, or by `typeid(my_variable)` in the extreme cases.

sherm1, 2019-03-04:
Sounds reasonable. We can deal with this case by case to see whether we have any actual problems left. I'm not wild about changing dynamic_cast to static_cast even when we "know" we've got the right type. Would be good to have an extra assertion there, at least in Debug. Although I agree if the reasoning is local enough it's fine (though that's not super common).

jwnimmer-tri, 2019-03-04:
I agree that (1) is likely to be rare.  I'd also argue that if the purpose is an assertion, there is a chance that writing it down as a `DRAKE_ASSERT` ends up being clearer than writing it as `unchecked_dynamic_cast`.

sherm1, 2019-03-04:
I like that -- we could just do that everywhere when we're worried about performance.
```c++
DRAKE_ASSERT(dynamic_cast<ConcreteType*>(&object_of_BaseType) != nullptr);
ConcreteType* concrete = static_cast<ConcreteType*>(&object_of_BaseType);
```

sherm1, 2020-05-12:
@jwnimmer-tri I believe you fixed this -- can we close it?

jwnimmer-tri, 2020-05-12:
That's a question for @amcastro-tri.

amcastro-tri, 2020-05-13:
It's been too long. We did push #10644 and probably other fixes. We'll need more benchmarks. I'll definitely get to benchmark a few things this quarter @sherm1. I particular, that sim you showed me with multiple falling objects. I'll see to reproduce and compare with Simobody's timings. Hopefully that'll tell us more about 1) algebra performance in general, 2) if overhead due to things like dynamics casts, 3) performance of simple error-controlled explicit Euler. 

Until then, I have not other measurements.

jwnimmer-tri, 2022-03-13:
Given the lack of any follow-up, I think we can close this now?  If we write more benchmarks that show hot spots, we can open new issues for those new problems?

closed 2022-03-14 by sherm1