RobotLocomotion/drake #10641: Write unit tests that ensure multibody code does not do any dynamic memory allocation.

amcastro-tri, 2019-02-08:
We want to ensure that many of our building blocks in the multibody engine do not do any dynamic memory allocation after a context is created, for performance reasons.  
For now this issue tracks:
1. Inverse dynamics doesn't do any dynamic memory allocation. Unit tested.
1. Mass matrix doesn't do any dynamic memory allocation. Unit tested.

from f2f @jwnimmer-tri, you mentioned you have ideas and even sample code for this?

amcastro-tri, 2019-02-08:
FYI, for inverse dynamics only 5% of the cost goes to malloc/free calls, so not a real bottleneck but definitely nice fixing.

jwnimmer-tri, 2019-02-10:
@amcastro-tri please try out #10650 and see if it's useful for this purpose.

jwnimmer-tri, 2019-02-12:
> FYI, for inverse dynamics only 5% of the cost goes to malloc/free calls, so not a real bottleneck ...

The usual complaint with mallocs is less that they degrade the mean throughput, but rather that they can introduce relatively high jitter for operations' latency, especially as system load increases.  Code in Drake that is going to be used for real-time control should (ideally) be written to avoid allocations after some start-up phase.

sherm1, 2020-05-12:
Related: #13186

closed 2024-03-20 by jwnimmer-tri