RobotLocomotion/drake #10644: Caching performance overhead.

amcastro-tri, 2019-02-08:
I am in the process of adding new cache entries to speed up the computation of the mass matrix. Lo and behold, I do get a nice performance boost as expected. However, the performance of our inverse dynamics gets affected! That is, caching has measurable overhead. I was careful not to do `EvalFoo()` within performance critical loops but instead to `Eval()` outside loops and pass the updated cache entries deep into the MBT code. Even being that careful, the performance hit was measurable, 10% to be more specific.

I did the following test; within `MultibodyTreeSystem` I swapped all the  `CacheEntry::Eval<FooCacheType>()` by `CacheEntry::UnsafeEval<FooCacheType>()`, where the latter is a hacky implementation without checks like so:
```c++
  template <typename ValueType>
  const ValueType& UnsafeEval(const ContextBase& context) const {
    const CacheEntryValue& cache_value = get_cache_entry_value(context);
    if (cache_value.needs_recomputation()) {
      UpdateValue(context);
    }
    const AbstractValue& abstract_value = cache_value.get_abstract_value();
    return static_cast<const Value<ValueType>*>(&abstract_value)->get_value();
  }
```
that is, `MaybeGetValue()` seems to be more expensive that what we'd like to believe. That was only responsible for about 7% of that cost. I couldn't tell where the other 3% is going.

sherm1, 2019-02-09:
Oh, you probably won't be surprised to see MaybeGetValue()'s first line:
```
    const Value<T>* value = dynamic_cast<const Value<T>*>(this);
```

Switching that to `unchecked_dynamic_cast` should make that very cheap (in Release builds).

jwnimmer-tri, 2019-02-09:
@amcastro-tri Could you check what difference #10649 makes to your numbers?

amcastro-tri, 2019-02-11:
Thanks @jwnimmer-tri, I'll rerun the tests.

sherm1, 2020-04-20:
I believe this is fixed now -- can we close it @amcastro-tri ?

amcastro-tri, 2020-04-20:
Definitely #10649 was an improvement, I haven't observed this overhead in any of my recent profiles.

closed 2020-04-20 by amcastro-tri