RobotLocomotion/drake #10645: systems: Non-owning `FixInputPort`?

EricCousineau-TRI, 2019-02-09:
Relates #10592

For some of my workflows, it's nice to have a non-Diagram built System, and just mock the inputs to quickly test the input, then throw the system away (or leave it completely decoupled).

This is nice because for MBP+SG workflows, adding any system to the diagram requires exclusive ownership transfer, meaning you can only do it once. It's nice to quickly test stuff without having to own stuff.

In Anzu, I'm porting stuff from RBT to MBP.
One thing I had before was a simple wrapper around the RBT `RgbdCamera` that handled all the systems stuff, so I could just write something like:
```
color, depth, label = SimpleCamera(tree, frame).render(q)
```
and have `np.ndarray` images.

Right now, I'm seeing how much I can avoid needing to transfer ownership, and am hacking around. I was _almost_ there, with (some hacky) code like this:
```py
class WrapperCamera(object):
    def __init__(self, source_id, scene_graph, camera):
        ...

    def render(self, sg_context):
        ...
        rsg_context = render_sg.CreateDefaultContext()
        poses = scene_graph.get_source_pose_port(
            self._source_id).EvalAbstract(sg_context)
        rsg_context.FixInputPort(
            render_sg.get_source_pose_port(self._source_id).get_index(),
            poses.Clone())
        c_context = camera.CreateDefaultContext()
        camera_context.FixInputPort(
            camera.query_object_input_port().get_index(),
            # Tip the domino...
            render_sg.get_query_output_port().EvalAbstract(rsg_context).Clone())
        # Render.
        # FAIL: QueryObject is (quite correctly?) non-copyable, and triggers
        # failure.
        color_image = camera.color_image_output_port().Eval(c_context)
```

Since this is throwaway objects, it'd be nice to temporarily affix a non-owned value to the input port.

The workaround, since luckily this is localized to `(dev::SceneGraph, RgbdCamera)`, is to put them both in a diagram.

However, this workaround will evaporate once `SceneGraph` gains rendering super powers.

@sherm1 @SeanCurtis-TRI @jwnimmer-tri Can I ask your thoughts on this?

jwnimmer-tri, 2019-02-09:
I don't understand the problem yet.  In C++, we have `ContextBase::FixInputPort(int index, const AbstractValue& value)` which copies the AbstractValue -- no lifetime issues at all.  Is this just that some pydrake binding is broken?

An AbstractValue _must always be cloneable_ full stop.  It sounds to me like either QueryObject is broken, or the SceneGraph is broken, or the pydrake bindings for them are broken (missing `Value<>` specialization in pydrake?) or something.  But I don't think ownership of FixInputPort is the problem, at least not that I can tell from your example.

EricCousineau-TRI, 2019-02-09:
I guess in this case, then, it's the fact that `QueryObject` is not cloneable. I can't pass it to `FixInputPort(const AbstractValue&)`, because `QueryObject::Clone()` will invalidate it's state:
https://github.com/RobotLocomotion/drake/blob/3dead00cf096389dd8473f7d9ef576f3d802ef9f/geometry/dev/query_object.h#L286-L287

Will rephrase title.

SeanCurtis-TRI, 2019-02-09:
`QueryObject`  is incomplete. The *actual* copy logic hasn't been implemented yet. It has a cloneable interface but it's one of those "I  promise to give you a clone interface if you promise not to call it." It wouldn't be too hard to implement it -- it, essentially, creates a copy of all of the state with fully updated poses so that it is no longer connected to the original context/system.

So, this may be the weight on the scale that says it's finally time to finish `QueryObject` -- up to now, people haven't tried copying so the dirty little secret has stayed in the bottom of the sock drawer.

EricCousineau-TRI, 2019-02-09:
That'd be excellent!
Can share more about this use case either on Monday or Tuesday.

SeanCurtis-TRI, 2019-05-16:
FYI PR #11484 is closing the gap on the `QueryObject` making a useful copy.

EricCousineau-TRI, 2019-05-16:
Aside from de-forking `dev/scene_graph`, is that PR pretty much all that is needed?

When "baking", what would it copy in the case of having `RenderEngine`s registered? Would it just copy an alias the `RenderEngine`?

SeanCurtis-TRI, 2019-05-16:
Nope. Deep copies all around; it can't be interfered with by the context.

EricCousineau-TRI, 2019-05-17:
Aye. At some point, for my use case, I may want to add in copy-on-write semantics for these more expensive parts, like `RenderEngine`. Will wait until it's an acute pain point, and then I'll pursue it (if there are no other amenable solutions).

SeanCurtis-TRI, 2019-05-17:
> Aside from de-forking dev/scene_graph, is that PR pretty much all that is needed?
 
I realize I forgot to answer this question.  :-/  I believe so. Obviously, I only applied the change on the mainstream `QueryObject`. It'd be nice not to have to duplicate it into `dev`. However, if you didn't want to wait for the deforking, duplicating would suffice.

SeanCurtis-TRI, 2019-07-30:
@EricCousineau-TRI I'm going to close this. Between #11484 and the imminent deletion of `geometry/dev` (and current evidence that suggests you can do this in the example you outlined above), it feels like it's time to close this.

As for the "copy-on-write" semantics you mooted, I think that may be more problematic than you think. However, that probably goes in its own issue.

Let me know if you agree with closing this and I'll close it (or you can).

EricCousineau-TRI, 2019-07-30:
Confirmed that this now works in Anzu, so yup, I'll go ahead and mark this as resolved by #11484 - thanks!

FTR, this is the `WrappedCamera` code that works (it's more-or-less a hack, but permits a non-Diagram-based connection between the `SceneGraph` and the instantiated camera):
```py
class WrappedCamera(object):
    """Wraps a `RgbdSensor` from Drake to more easily render and retrieve
    images. This permits instantiating a camera that is not registered in a
    diagram which contains the given scene graph."""

    def __init__(self, scene_graph, camera):
        """
        @param scene_graph SceneGraph instance
        @param camera RgbdSensor instance (will take ownership).
            This should NOT be owned by another diagram, nor be connected.
        @pre scene_graph Has a renderer registered that corresponds to the
            properties described by `camera`.
        """
        # TODO(eric.cousineau): Use `SceneGraph.HasRenderer` to explicitly
        # check this.
        assert isinstance(scene_graph, SceneGraph)
        assert isinstance(camera, RgbdSensor)
        self._scene_graph = scene_graph
        self._camera = camera
        self._context = self._camera.CreateDefaultContext()
        # Easier way to get an input port's model value?
        self._fixed = None

    def get_camera_info(self, camera_type):
        if camera_type == "color":
            return self._camera.color_camera_info()
        elif camera_type == "depth":
            return self._camera.depth_camera_info()
        else:
            raise ValueError("Unknown camera {}".format(camera_type))

    # TODO(eric.cousineau): Use actual port names?
    def render(self, sg_context, calc=['color', 'depth', 'label']):
        """Renders image given scene graph context.
        @param sg_context SceneGraph context.
        @return Image for each type specified in `calc`.
        """
        # Allocate contexts, mock connections.
        query = self._scene_graph.get_query_output_port().EvalAbstract(
            sg_context)
        if self._fixed is None:
            # N.B. This is *very* expensive, in that it will copy all of the
            # values of the QueryObject (including render + proximity data).
            self._fixed = self._camera.query_object_input_port().FixValue(
                self._context, query)
        else:
            self._fixed.GetMutableData().SetFrom(query)
        # Render.
        context = self._context
        out = list()
        for image_type in calc:
            if image_type == "color":
                image = self._camera.color_image_output_port().Eval(context)
                out.append(np.copy(image.data[:, :, :3]))  # No alpha
            elif image_type == "depth":
                image = self._camera.depth_image_output_port().Eval(context)
                out.append(np.squeeze(np.copy(image.data)))
            elif image_type == "label":
                image = self._camera.label_image_output_port().Eval(context)
                out.append(np.squeeze(np.copy(image.data)))
            else:
                raise ValueError("Unknown image type: {}".format(image_type))
        return out
```

closed 2019-07-30 by EricCousineau-TRI