RobotLocomotion/drake #10647: systems: Move API to use `InputPort`, `OutputPort`, instead of indices?

EricCousineau-TRI, 2019-02-09:
At present, things like `Context.FixInputPort()` and `DiagramBuilder.ExportInput()` interface with indices.

It'd be nice if they all just used the first-class objects, consistently. That way, it'd motivate users to more closely inspect their methods, and learn about `Eval()` as it'd be introduced in #10592.

\cc @sherm1 @jwnimmer-tri

jwnimmer-tri, 2019-02-09:
There a question of whether Context is supposed to know about System (or InputPort etc).  I think so far it does not -- the layering is that Systems depend on Contexts, but not the other way around.  Maybe we should change that, but should be careful if we do.

For this specific request though, we could offer a

```c++
FixedInputPortValue& InputPort<T>::Fix(Context*, const AbstractValue&)`
```

instead, perhaps?  (And of course the Eigen sugar as well.)  So it would be used like:

```c++
camera.pose_input_port().Fix(&context, my_q);
```

(Though per #10516 we might want to re-evaluate the verb "Fix".)

EricCousineau-TRI, 2019-02-09:
Yup, that'd look better!

What might it look like for `DiagramBuilder::Export*`?

jwnimmer-tri, 2019-02-09:
Those already take Port objects, so are good?  Oh, I see -- you mean the return type, since we don't have a System yet we can't return an InputPort.  So the question you're asking is how do you get, say, an InputPort object for a Diagram once it's built.

I guess the answer for now is that you can just grab them by name once you're done?  (Or that you are is-a Diagram, in which case you can make methods for them.)

But really my answer is that DiagramBuilder is bad, and we should be able to add+remove systems and ports and connections and fixed values and etc willy nilly, anytime we want (death to Finalize / Build) and so in that case we would return the Port object right away when we added it.

jwnimmer-tri, 2021-11-11:
We have `InputPort.FixValue` now.  Absent any more specific complaints, this is finished.

closed 2021-11-11 by jwnimmer-tri