RobotLocomotion/drake #10651: Support for unprovisioned python3 builds

jwnimmer-tri, 2019-02-11:
In https://drake-jenkins.csail.mit.edu/view/Unprovisioned/, I don't see any `-python3` options.  Could we add some?  We probably don't need them all, but at should have at least the one pre-merge required build (`linux-bionic-gcc-bazel-experimental-python3-release`) in unprovisioned form.

Every once in a while, a PR might need to add some python3-specific new dependency (see, e.g., #10632 recently).  It would be nice to have the ability to test a clean Drake build + the new packages, prior to merge (and AMI updates).

\CC @EricCousineau-TRI FYI

jamiesnape, 2019-02-11:
We don't actually have unprovisioned builds for most configurations. I am not saying we shouldn't, but it was a deliberate omission since we are actually constrained on the number of experimental jobs courtesy of some design decisions in the Jenkins plugin and the GitHub rate limit.

jamiesnape, 2019-02-11:
(But anyway, I will work on getting at least one in)

jamiesnape, 2019-02-11:
Bionic Unprovisioned Python 3 builds have been created.

closed 2019-02-11 by jamiesnape