RobotLocomotion/drake #10654: MBP::CalcJacobianSpatialVelocity doesn't compute the gradient ∂ ᴬṗᶜ / ∂ q̇ when the quaternion is unnormalized

hongkai-dai, 2019-02-11:
If I want to compute the partial gradient `∂ ᴬpᶜ / ∂ q = ∂ ᴬṗᶜ / ∂ q̇` for a point `C`'s spatial velocity measured and expressed in `A`'s frame, w.r.t `q̇`, I would like to call In `MBP::CalcJacobianSpatialVelocity`. This function gives me the right gradient when the quaternion in `q` is normalized. But when it is not normalized, this function doesn't compute the partial gradient. It actually computes the gradient w.r.t the normalized quaternion `q̃`, namely `∂ ᴬpᶜ / ∂ q̃`. The gradient I hope to get is ∂ ᴬpᶜ / ∂ q = ∂ ᴬpᶜ / ∂ q̃ *  ∂ q̃ / ∂ q `.

cc @gizatt this is the gradient problem you observed last week, when you want to impose collision avoidance constraint between box-sphere or box-box.

cc @avalenzu

sherm1, 2019-02-11:
Per f2f with @hongkai-dai @mitiguy @amcastro-tri:

The problem here appears to be that the kinematic calculation of point location from q (or velocity from q and qdot) is not consistent with the Jacobian calculation, with respect to treatment of unnormalized quaternions.

I feel strongly that the correct behavior when given unnormalized quaternions is to normalize them _internally_ (so that we don't get non-orthogonal rotation matrices that produced illusory constraint violations). Our RotationMatrix class does that very efficiently. But then the Jacobian needs to be the partial derivative of _that_ kinematic calculation, not the unnormalized one.

hongkai-dai, 2019-02-11:
> The problem here appears to be that the kinematic calculation of point location from q (or velocity from q and qdot) is not consistent with the Jacobian calculation, with respect to treatment of unnormalized quaternions.

That is right, we have two functions. `CalcPointsPosition`, which computes the position `ᴬpᶜ` of a point `C`, attached to a body `B`, measured in a frame A; and a function `CalcJacobianSpatialVelocity`, which I think should compute the gradient `∂ ᴬpᶜ / ∂ q`.

What these two functions currently do in Drake is this:

1. For `CalcPointsPosition`, it first normalizes the quaternion `q` *internally* to get `q̃ = q / |q|`, and then it computes the point position using this normalized `q̃`. So the computation always uses orthogonal rotation matrices.
2. For `CalcJacobianSpatialVelocity`, it also first normalizes quaternion `q` *internally* to get `q̃ = q / |q|`, and then it computes the Jacobian using this normalized `q̃`, hence the Jacobian is computed as `∂ ᴬpᶜ / ∂ q̃`. But I think this computation misses the part on `∂ q̃ / ∂ q`, which is the gradient coming from the normalization. We should multiply this gradient `∂ q̃ / ∂ q` back to `∂ ᴬpᶜ / ∂ q̃`, to get the gradient w.r.t the unnormalized quaternion `q` as `∂ ᴬpᶜ / ∂ q = ∂ ᴬpᶜ / ∂ q̃ * ∂ q̃ / ∂ q`

sherm1, 2019-02-11:
With @hongkai-dai's clarification I _think_ this is just a bug -- the Jacobian function is not actually returning exactly the derivative of the kinematic function we're actually using. BTW if `q̃ ≜ q / ‖q‖`, then 
```
∂ q̃   I - q̃q̃ᵀ
--- = ------
∂ q    ‖q‖
```

hongkai-dai, 2019-02-12:
I have a [dev branch](https://github.com/hongkai-dai/drake/tree/unnormalized_quaternion_jacobian) to reproduce the problem. In this [test](https://github.com/hongkai-dai/drake/blob/unnormalized_quaternion_jacobian/multibody/inverse_kinematics/test/jacobian_test.cc) , it computes the Jacobian for two different configurations, one with a normalized quaternion, the other configuration is the same as the first one, except its quaternion is twice as large. I compute the position of a point `Q` fixed to frame `S`, measured in frame `B` using both quaternions. Also I compute the Jacobian using either `CalcJacobianSpatialVelocity`, or `MBP<AutoDiffXd>`. Here is the result
```
p_BQ_normalized: 0.91358 2.65432 4.69136

p_BQ_unnormalized: 0.91358 2.65432 4.69136

Jq_v_BQ_normalized:
  1.26224   9.75983  -3.65148  -2.45686  0.666667 -0.666667 -0.333333 -0.315846  -6.64285   2.27141   2.29493 -0.666667  0.666667  0.333333
 -2.09171  0.712265   7.51935  -5.47272 -0.133333  0.333333 -0.933333   1.13167  0.100802  -5.17585   3.62753  0.133333 -0.333333  0.933333
 0.937665  -2.30359  -3.54329   3.57485 -0.733333 -0.666667 -0.133333 -0.630349   1.15922   2.73106  -2.62825  0.733333  0.666667  0.133333

Jq_v_BQ_autodiff_normalized:
  1.26224   9.75983  -3.65148  -2.45686  0.666667 -0.666667 -0.333333 -0.315846  -6.64285   2.27141   2.29493 -0.666667  0.666667  0.333333
 -2.09171  0.712265   7.51935  -5.47272 -0.133333  0.333333 -0.933333   1.13167  0.100802  -5.17585   3.62753  0.133333 -0.333333  0.933333
 0.937665  -2.30359  -3.54329   3.57485 -0.733333 -0.666667 -0.133333 -0.630349   1.15922   2.73106  -2.62825  0.733333  0.666667  0.133333

Jq_v_BQ_unnormalized:
  2.52448   19.5197  -7.30297  -4.91373  0.666667 -0.666667 -0.333333 -0.315846  -6.64285   2.27141   2.29493 -0.666667  0.666667  0.333333
 -4.18343   1.42453   15.0387  -10.9454 -0.133333  0.333333 -0.933333   1.13167  0.100802  -5.17585   3.62753  0.133333 -0.333333  0.933333
  1.87533  -4.60718  -7.08658    7.1497 -0.733333 -0.666667 -0.133333 -0.630349   1.15922   2.73106  -2.62825  0.733333  0.666667  0.133333

Jq_v_BQ_autodiff_unnormalized:
 0.631121   4.87991  -1.82574  -1.22843  0.666667 -0.666667 -0.333333 -0.315846  -6.64285   2.27141   2.29493 -0.666667  0.666667  0.333333
 -1.04586  0.356132   3.75968  -2.73636 -0.133333  0.333333 -0.933333   1.13167  0.100802  -5.17585   3.62753  0.133333 -0.333333  0.933333
 0.468832   -1.1518  -1.77165   1.78742 -0.733333 -0.666667 -0.133333 -0.630349   1.15922   2.73106  -2.62825  0.733333  0.666667  0.133333
```

As you can see, that the position `p_BQ` are the same, for either the normalized quaternion and the unnormalized quaternion, indicating that when we compute the position in `CalcPointsPosition`, it internally normalizes the quaternion. But when we compute the Jacobian, the unnormalized quaternion gives a Jacobian that is twice as large as the normalized one (in the first 4 columns corresponding to the quaternion), while the actual answer should be one half (as the result using AutoDiffXd). This indicates that the function `CalcJacobianSpatialVelocity` doesn't do any internal normalization. It just takes any quaternion, whether normalized or not, as @sherm1 pointed out.

Hence there is a mismatch between `CalcPointsPosition` (which does quaternion normalization internally) and `CalcJacobianSpatialVelocity` which doesn't normalize the quaternion. I would expect these two functions to be consistent, that either they both do the normalization internally, or neither does normalization. Personally I prefer that they both normalize the quaternion internally.

PR #10671 (2019-02-13): Updates computation of Nplus(q) for non-unit quaternions.
Resolves #10654.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10671)
closed 2019-02-13 by amcastro-tri