RobotLocomotion/drake #10656: rbp: Failures in LCP solves

EricCousineau-TRI, 2019-02-11:
@mpetersen94 and @neeld are running into an issue where RBP's LCP solve is failing in C++, and sometimes in Python (depending on which commit they're using).

Working to generate min repro.

EricCousineau-TRI, 2019-02-11:
Min repro (for failure in C++ and Python):
https://github.com/EricCousineau-TRI/drake/blob/issue/10656_repro-wip/tmp/BUILD.bazel

mpetersen94, 2019-02-18:
I think the problem appeared as a result of commit [e976861](https://github.com/mpetersen94/drake/commit/e976861021979fa576a8560e29775c1f5d5688de) because running a null simulation of my RigidBodyPlant with closed loop works when using a commit of drake right before it ([mpetersen94/m-328](https://github.com/mpetersen94/drake/tree/m-328)) but fails when using a commit of drake right after it ([mpetersen94/m-327](https://github.com/mpetersen94/drake/tree/m-327)).

sherm1, 2019-02-18:
cc @edrumwri -- that commit is the infamous #9766 to handle events at initial time step. Not clear how that could be related to an LCP failure though?

jwnimmer-tri, 2019-02-18:
@sherm1 Recall that #9766 broke master pydrake on macOS when it landed, because of floating point denormalization.  One question is whether cherry-picking the repair in #10260 (598e3bb93b7263e5185a637025a8cbd048127794) on top of m-328 resolves this problem.  Also of interest is whether `simulator.set_publish_at_initialization(false)` (to opt-out) would also resolve it, or whether perhaps solving the LCP at `time < 0` (the time warp implemented by Initialize) is a problem.

mpetersen94, 2019-02-18:
Cherry-picking the repair from #10260 on top of m-237 (the failing branch) did not fix the problem.  I also tried using `simulator.set_publish_at_initialization(false)` as well as commenting out simulator.Initialize() and neither of these resolved the problem.

edrumwri, 2019-02-19:
@mpetersen94 investigating this is in my queue but might not happen until tomorrow. Thanks for your patience!

edrumwri, 2019-02-20:
@mpetersen94 using @EricCousineau-TRI's reproduction code (thanks Eric!) and git bisection, PR #10262 (2a0d9a2) which disabled the -ffast-math option for Ospray, was the responsible party. This is not surprising, as my analysis of your problem indicated a floating point problem and -ffast-math mucks with the FPU state  _just by being present in the shared library_!

The code is actually correct now; the discussion in #10262 goes into some detail about this. Our hypothesis was that there was a denormalized float which -ffast-math was causing (fortuitously in your case) to treat as zero. Indeed, if you put this code at the beginning of your main(.) function, it will "fix" it.

    const unsigned MXCSR_DAZ = (1 << 6);
    const unsigned MXCSR_FTZ = (1 << 15);
    const unsigned mxcsr = __builtin_ia32_stmxcsr() | MXCSR_DAZ | MXCSR_FTZ;
    __builtin_ia32_ldmxcsr(mxcsr);

I use the scare quotes around "fix" because there is an underlying problem that is not being addressed: you are not getting an accurate solution. Unfortunately, this is all we can offer. RigidBodyPlant is at the end of its maintenance cycle. MultibodyPlant has been designed from the ground up to supplant it because RBP was infeasible to maintain. MBP should have the features you seem to require (it's scheduled to land in Q2/Q3 of this year).

closed 2019-02-20 by edrumwri

sherm1, 2019-02-20:
@mpetersen94, if you want to debug this I'd suggest taking your big model and chopping it down to something small until it functions correctly, then add elements back piecewise until you can find the piece that creates a dependency on the behavior of denormalized numbers. Most likely this is due to some uninitialized data being referenced ... somewhere :( Conceivably a valgrind run could also reveal something.

edrumwri, 2019-02-20:
Just to clarify: we're not saying that this is a problem in your code. But
if you need to debug the Drake code, @sherm1's suggestion is where I'd
start.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA