RobotLocomotion/drake #10664: The documentation for Jacobian methods can be difficult to parse on a first pass.

amcastro-tri, 2019-02-12:
The current documentation for Jacobians has caused some headaches to expert roboticists not yet familiar with notation for frames used within Drake. Though complete, users looking for a quick answers to their needs struggle to find the bits of information they need from this complete description of frames.

For instance, a user might want to compute `Jdot`. Typically this is a Jacobian for a given frame (many times a body frame) and it is expressed in the world frame. Given these priors, the additional information encoded by Drake's notation is omitted by most users, and their notation simple reads `Jdot`. 

A possible solution could probably have an introductions on this matter in an `@name` section for the Jacobian methods. With this additional information in the `@name` section probably the per-method docs can be simplified. 

Another solution (maybe both are a good idea) is that the first "brief" description of a Jacobian methods is simply stated in words. For instance, for `CalcBiasForFrameGeometricJacobianExpressedInWorld()` we could say something like:

"This method computes the bias term Ab = J̇⋅v for the spatial acceleration A of a frame F, which can be expressed as A = J⋅v̇ + Ab with J the Jacobian corresponding to frame F. To be more precise .... **here we continue with the more complete description including Drake's notation for frames.**"

cc'ing @mitiguy, @sherm1, @edrumwri, @avalenzu, @mposa

sherm1, 2019-02-12:
:+1: great idea, @amcastro-tri . @mitiguy can you incorporate this in your current Jacobian doc revamp?

amcastro-tri, 2019-02-13:
from f2f discussion with @mitiguy. He's working on a great pdf document with notation and equations. Probably something really good to incorporate to our Drake notation Doxygen page. Then it'd be a matter of referencing that section from the Jacobian methods.

sherm1, 2020-05-12:
I think this is done now, @mitiguy and @amcastro-tri ? Can we close the issue?

amcastro-tri, 2020-05-13:
Great reminder @sherm1, thanks. Essentially we've been reminding ourselves to do so per each PR on this Epic. We easily drift towards writing lengthy documentation that nonone reads unfortunately. Thus far the solution we've tried with @mitiguy is to have the first (2-3 lines) paragraph give the short roboticist version. Then expand in subsequent paragraphs for those interested. It seems a good trade off. @mitiguy, what do you think?

mitiguy, 2020-05-21:
This issue is closed as issue #12140 has been addressed with PR# 13036.
That issue and PR have to do with rename, notation, documentation, and code for CalcBiasSpatialAcceleration and CalcBiasTranslationAcceleration().

closed 2020-05-21 by mitiguy