RobotLocomotion/drake #10675: Unfork the SceneGraph

jwnimmer-tri, 2019-02-13:
I spent a few minutes searching the issues database, but I didn't find a tracking issue opened for this yet.  Please feel free to resolve this as a duplicate if I missed it.

We have some code in `geometry/` and a fork of that code in `geometry/dev/`.  There should be only a single copy of our latest-and-greatest software (on master), both so that users have a single thing that just works, so that Drake maintainers only have to update one copy of the code with bugfixes, API evolution, bindings, etc., and for all of the other usual reasons why forking is bad.

I'm filing this issue now so that we can triage its priority at the weekly issue scrub.  My opening bid is that we should add a single new feature or bugfix in `geometry/` again until this is resolved, if there is any way the person touching the code could instead be working on unforking it.

SeanCurtis-TRI, 2019-02-13:
Several thoughts:

1. I've updated #9540 to explicitly include removing `geometry/dev`. That also outlines the prerequisites for that happening.
2. I think your opening bid is not practical in the current environment (although it has long been my preference). Given the contents of #9540 there are still multiple PRs before `geometry/dev` can go away. and there are numerous of high-priority interruptions. That said, if we can decide that my highest priority is always #9540 and everything else merely gets to fill the cracks, then, at the very least, I can make progress.
3. Also, there's no reason to limit changes in the proximity engine while `geometry/dev` still lives. The two are largely orthogonal in that respect and nothing done in `geometry/` for proximity work needs to be reflected in `geometry/dev` at all.

jwnimmer-tri, 2019-02-13:
> ... there's no reason to limit changes in the proximity engine while `geometry/dev` still lives.

Right.  My reasoning was more about staffing than code dependencies -- an "all hands on deck" or "fire drill" mentality.  If we can put six people on this and get it done in two weeks, we should.  (Though of course it _probably_ doesn't parallelize to six people, but we should see how high it can get.)  I'd be willing to sacrifice efficiency to get more people doing these PRs in the name of latency.  It shouldn't have to be all on Sean.

> ... outlines the prerequisites for that happening.

That list as I see it now seems to imply that RGB camera rendering must exist in non-dev SG before we can kill off dev SG.  I would like us to revisit that assumption.  I think that once we have "Add `RenderEngine` abstraction to `geometry`" that unblocks the dataset generation work (where the user writes their own renderer) then there is no high-priority need for RGB camera rendering.

> ... If we can decide that my highest priority is always #9540 and everything else merely gets to fill the cracks ...

Yup, in the scrub meeting I will try to articulate why other people's high priority items should come after this issue.  And per above, I think this issue should be _multiple people's_ highest priority.  I am happy to include myself in that list as well.

SeanCurtis-TRI, 2019-02-13:
We should probably talk briefly f2f. I think there are some assumptions we're not sharing. In principle I'm fully on board and greatly appreciative and I think this is a push in the right direction. In practice, the ideas of highly-parallelizing #9540 or whether or that items on the list are not actually pre-requisites is a total disconnect for me.

jwnimmer-tri, 2019-02-13:
Sure, I am happy to wait until the scrub meeting, or else to do a VTC earlier if that helps.

Maybe my TL;DR to your question is that I think we should **not** implement rendering **at all** right now in Drake other than a minimal API hook to write an offline dataset renderer in Anzu, and that we should lobotomize the manipulation station so that it doesn't have RGB cameras anymore so that we can remove dev SG.  Once all that is done, and other high priorities like depth / collision are taken care of, we can come back and work to get rendering added to SG.

SeanCurtis-TRI, 2019-02-13:
I'm available for chatting whenever. If you think it will contribute to the scrub meeting, then before. If not, whenever.

SeanCurtis-TRI, 2019-03-07:
See #9540 for update

SeanCurtis-TRI, 2019-04-29:
Previous comment still applies: See #9540 for status.

jwnimmer-tri, 2019-04-29:
Not a week goes by where I don't burn at least 1-2 hours dealing with `dev` being a fork -- where I have to edit, compile, test, deprecate etc. a second copy of the code.  I really wish we could get an MVP shipped of whatever feature is blocking killing off dev (I guess, render something something) without so much complexity, so that we can kill off dev ASAP.

tri-ltyyu, 2019-05-09:
Closing this so we can track as part of #9540

closed 2019-05-09 by tri-ltyyu