RobotLocomotion/drake #10676: cc failed ubuntu

enriquetrucios, 2019-02-13:
> Before you post:

> * Have you reviewed the documentation on
[Getting Help](https://drake.mit.edu/getting_help.html)?
> * Have you reviewed existing GitHub issues and StackOverflow posts?
> * Have you rerun `install_prereqs.sh` on your current commit of `drake`?

> If you have encountered a bug (installation, segfaults, etc.), please try to
post reproduction steps and as much relevant output as possible. Consider
posting long snippets as a [Gist](https://gist.github.com/).

jwnimmer-tri, 2019-02-13:
Were there any other parts of the error output that were interesting?

My best guess at the moment is that kernel killed the compiler due to memory pressure, or that the computer ran out of disk space.  How many cores and how much RAM does this computer have?

To reduce memory pressure, you can reduce the number of jobs the build tries to run at once -- create the file `$HOME/.bazelrc` and with the line `build --jobs=2` (or some other number).  (Unfortunately `make -j2` doesn't percolate through to the underlying Bazel build system -- by default, it tries to use almost all memory and processors, and sometimes it is too aggressive.)

enriquetrucios, 2019-02-14:
Effectively, the kernel had killed the compiler.
The procedure increased by 2 hours but It didn't make any mistakes.
Thanks a lot.

closed 2019-02-14 by enriquetrucios