RobotLocomotion/drake #10692: Add sugar for fixing input port value

sherm1, 2019-02-14:
(From discussion with @mposa in PR #10685.) For the use case where actuation inputs u are to be provided directly rather than via connected port, the current FixInputPort incantations were confusing. Proposal:

- Make sure the MBP actuation ports are always present, even when there are zero actuators.
- Add a method like `InputPort::FixValue(context, value)`.

This would allow
```c++
plant.get_actuator_port().FixValue(&context, u);
```

To me that seems like a nice follow-on to Jeremy's #10592 that allows 
```c++
u = plant.get_actuator_port().Eval(context);
```

Thoughts?

/cc @edrumwri @amcastro-tri @jwnimmer-tri

sherm1, 2019-02-14:
/cc @rcory due to your recent battles with FixInputPort.

edrumwri, 2019-02-14:
I like both of these proposals, but would propose separating the issues.

sherm1, 2019-02-14:
> ... would propose separating the issues.

Good idea - #10694 for zero-length ports. Will leave this issue for the InputPort API addition.

rcory, 2019-02-14:
I think this is a great proposal. My one request is to make sure that the documentation for `InputPort::FixValue` calls out the "stomping" effect it will have on existing connections (via the underlying call to `FixInputPort`). Easiest thing would be to copy the comment from `Context::FixInputPort` (see #10665).

jwnimmer-tri, 2019-02-14:
My one request is to avoid some of the poor argtype choices of Context::FixInputPort, and instead only offer three options that mirror what `InputPort::Eval` offers:
```c++
class InputPort<T> {
  ...
  FixedInputPortValue& FixValue(Context<T>*, const Eigen::Ref<const VectorX<T>>& vector) const;

  template <typename U>  // + SFINAE that Value<U> is valid; bonus for automatic upcasting
  FixedInputPortValue& FixValue(Context<T>*, const U& value) const {
    FixValue(context, Value<U>(value));
  }

  FixedInputPortValue& FixValue(Context<T>*, const AbstractValue& abstract_value) const;
...
}
```

Remember that the argument names are important in pydrake to force a sensible overload using kwargs when necessary.  (I took my best shot above.)

sammy-tri, 2019-02-19:
Assigning to @sherm1 as default owner.

sherm1, 2019-03-06:
Status: PRs #10759 and #10838 in review.

closed 2019-03-20 by sherm1

sherm1, 2019-03-20:
@mposa, your idea for simplifying actuator-setting for MultibodyPlant (after much mutation and haggling and three PRs) is now in Drake. Please give it a try and let us know whether you think it simplifies the API enough to be reasonable. You can now write:
```c++
plant.get_actuator_port().FixValue(&context, value);
```
where `value` is a vector of the right size (and zero is OK). It's not quite as nice as `SetActuator()` would have been, but this is fully general and works for any kind of port anywhere.

PR #10759 (2019-03-20): Add FixValue method to InputPort
Adds an `input_port.FixValue(&context, value)` method using the new ValueToAbstractValue & ValueToVectorValue policy classes from #10838.

This is a replacement for the uglier `context.FixInputValue(port_num, value)` method collection which caused @mposa some usability headaches.  

Resolves #10692.

```
Category            added  modified  removed  
----------------------------------------------
code                153    31        9        
comments            77     21        10       
blank               39     0         0        
-------------------------- [... cut]