RobotLocomotion/drake #10700: implement "static stability" constraint for MBP

RussTedrake, 2019-02-15:
and use it in the manipulation station clutter clearing demo to project the random initial conditions out of penetration and into (approximate) contact solutions.

My proposal is to write, e.g.:

```
min_{q,lambda)  (q - q_sample)^2

s.t. 0 = τ_g(q) + B*u + J^T(q) λ 
     λ_i = linearized_soft_contact_model(signed_contact_distance_i(q))
     signed_contact_distance_i(q) * λ_i <= 0  
     + similar for friction
```
where the middle line is meant to be like the soft contact version of the complementarity (when combined with the equality constraint).

My thinking is that the numerics will be much better using a simplified soft contact model (e.g. instead of the stiff stribeck model), and that having lambda as an explicit decision variable will inform the solver that it can pull things into contact in order to satisfy the force balance.

though I'm open to variations on it.

cc @hongkai-dai

hongkai-dai, 2019-02-15:
May I ask how the user should call this "static equilibrium solver"?  I suppose the author has an MBP and a SceneGraph for the collision query. Do the authors know what are the candidate contact bodies? Otherwise for each pair of bodies in the collision geometry, we need to assign decision variable lambda to represent the contact wrench.

RussTedrake, 2019-02-15:
My use case is here: https://github.com/RobotLocomotion/drake/blob/master/examples/manipulation_station/manipulation_station.cc#L397

Yes -- I know the candidate contact bodies.

hongkai-dai, 2019-02-15:
I had a discussion with @sherm1, and here is a tentative plan

1. For each pair of contact geometries, create decision variables to represent the tangential friction forces. Here I assume each pair of contact geometries has only one pair of contact points (which is true when we only use sphere for contact, but would not be true in the future if we use box-box contact or mesh-mesh contact). The contact geometry pairs could be specified by the users, or the program just regard all pairs of geometries as candidate contact pairs (except for the pairs on the same rigid body).
2. Query the `ContactResult` from MBP/SG. This contains the normal force for each pair of contact geometries. I can then compute the Jacobian of the contact point. Also I can compute the normal contact force λₙ using the penetration depth.
3. The constraint I would like to impose is 
   ```
   λₙ,ᵢ = stiffness * penetration_depth(q)
   |λₜ,ᵢ| ≤ μ λₙ,ᵢ
   ```
   I don't think I have to explicitly add `λₙ,ᵢ` as decision variables? I think I can do |`λₜ,ᵢ| ≤ μ*stiffness * penetration_depth(q)`. BTW, I don't think I need the constraint `signed_contact_distance_i(q) * λ_n,i <= 0`? It is implied by computing normal force from the penetration depth already.

RussTedrake, 2019-02-19:
it all sounds great.

my suggestion to use the normal force as an explicit decision variable was only to "help" the solver by showing it a way to achieve force even if the nonlinear constraint coming from the penetration depth has no gradient.  (I wasn't happy with my formulation above in this regard yet, but wanted to put something down to start the discussion).

sort of interesting to consider
```
0 ≤ λₙ,ᵢ
λₙ,ᵢ (λₙ,ᵢ - stiffness * sdf(q)) ≤ 0
|λₜ,ᵢ| ≤ μ λₙ,ᵢ
```
that would be allow 0 ≤ λₙ,ᵢ ≤ - stiffness * sdf(q) during penetration and λₙ,ᵢ = 0 out of penetration, but have a gradient everywhere.  and the friction + plus resisting gravity would presumably push up on the normal force; under-estimating the normal force seems aok here.

hongkai-dai, 2019-02-19:
Thanks @RussTedrake 

I am not sure if I completely follow the formulation above. Particularly these constraints
```
     0 ≤ λ_{n,i}
     - stiffness * sdfᵢ(q) ≤ λ_{n,i}
     λ_{n,i} (λ_{n,i} - stiffness * sdfᵢ(q)) ≤ 0
```
From the first constraint `0 ≤ λ_{n,i}` and the third one `λ_{n,i} (λ_{n,i} - stiffness * sdfᵢ(q)) ≤ 0`, we know that  `λ_{n,i} - stiffness * sdfᵢ(q) ≤ 0` (when `λ_{n,i} > 0`), namely `λ_{n,i} ≤ stiffness * sdfᵢ(q))`. Combining this with the second constraint we get 
```
- stiffness * sdfᵢ(q) ≤ λ_{n,i} ≤ stiffness * sdfᵢ(q))
```
To make sure that `- stiffness * sdfᵢ(q) ≤ stiffness * sdfᵢ(q))` implied by the constraint above, we have to have `sdfᵢ(q) ≥ 0`, namely there can be no penetration. I don't think this is what we want? Maybe there is a typo in the constraint above, and a modified version could be ( I changed the `-` sign in the constraint ` λ_{n,i} (λ_{n,i} - stiffness * sdfᵢ(q)) ≤ 0` to `+`.
```
min_{q, λ}  (q - q_sample)^2

s.t. 0 = τ_g(q) + B*u + J_n^T(q) λ_n + J_t^T(q) λ_t
     0 ≤ λ_{n,i}
     - stiffness * sdfᵢ(q) ≤ λ_{n,i}
     λ_{n,i} (λ_{n,i} + stiffness * sdfᵢ(q)) ≤ 0
     | λ_{t,i} | <= μ λ_{n,i}
```
The admissible solutions are  `λ_{n,i} = 0, sdfᵢ(q)  >= 0` or `λ_{n,i} = - stiffness * sdfᵢ(q) , sdfᵢ(q) < 0`

RussTedrake, 2019-02-20:
i agree.  i had a typo.  thx.

hongkai-dai, 2019-03-19:
#10944 adds the static stability constraint g(q) + B * u + J' * lambda = 0. It doesn't add the friction cone constraint, or the complementarity constraint, since the friction cone constraint depends on the computing the contact normal, and the gradient of the contact normal w.r.t q.

PR #11400 (2019-05-12): Static equilibrium problem
Resolves #10700

This assumes Coulomb friction for rigid contact.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11400)
closed 2019-05-12 by hongkai-dai