RobotLocomotion/drake #10714: lcm_driven_loop: Simplify contract to permit more flexibility

EricCousineau-TRI, 2019-02-19:
At present, `LcmDrivenLoop`'s public contract is heavily bound to using a `LcmSubscriberSystem` to get its next update time.

This interface should be simplified to where the next update time is polled via something like a stateful `std::function<double ()> get_next_time_blocking`, s.t. you could then implement this for something like @mposa's use case (multiple LCM channels + non-LCM data), or other cases like ROS1, ROS2, etc.

Should be relatively low effort. My guestimate is being able to do this by end of next week.

jwnimmer-tri, 2019-02-19:
I would rather us remove `LcmDrivenLoop` entirely, than try to generalize it.  IMO it's too application-specific and too immature of a design to try to develop as part of the core library releases.

EricCousineau-TRI, 2019-02-19:
I'd be on board with that, but we should have something to capture that workflow (interfacing a diagram with externally driven time source).

sherm1, 2019-02-19:
@mposa is using LcmDrivenLoop currently -- although limited, I think it provides some useful functionality and we should be careful about replacing it to make sure it addresses actual user needs.

mposa, 2019-02-19:
I don't think it should be removed without being replaced--it represents pretty core functionality for synchronized code execution.

jwnimmer-tri, 2019-02-19:
But it's a bad design.  A user should be able to add an LcmSubscriberSystem to a diagram (marking it as a time source when creating the subscriber), as well as as many LcmPublisherSystems as they want, and then just call `Simulator::StepTo(infinity)` and it should all just work.  Trying to shoehorn that into a weird loop API that leads to weird bootstrapping copypasta at every call site is no good.

EricCousineau-TRI, 2019-02-19:
Hm... I'm not sure if I'm a fan of marking a system as a time source, as that would leak into the rest of the execution semantics and lead to even weirder edge cases in the event system than we already have. (I guess I'd be fine with this approach if we could eliminate *all* redundant event mechanisms, s.t. there's only one correct way to do a particular type of event.)

I'd much rather prefer something akin to `LcmDrivenLoop::StepTo`

Given this, though, my preference would be to still implement the simplification (at least to satisfy existing use cases), and then consider alternative / replacement as a follow-up, lower priority issue in which we can better hash out the design.

jwnimmer-tri, 2019-02-19:
I think trying to weave the "many possible sources of time that the loop consolidates" is much harder to get right than you are anticipating, and much much lower priority for our end-users than dozens of other enhancements that we could be pursuing in the meantime.

EricCousineau-TRI, 2019-02-19:
I really don't think it will be difficult; it really is just `std::function<double()>`. Lifetime will be the user's problem, not this interface's.

I set this time period b/c I think it should be easy; however, I am definitely up for timeboxing it, s.t. if it takes more than 30min, I punt on it and provide a non-`master` workaround.

EricCousineau-TRI, 2019-04-01:
@mposa You may want to take a look at Jeremy's WIP branch here:
https://github.com/RobotLocomotion/drake/pull/10981

This feature has the potential to simplify things *greatly*, and remove the need for complicated threading / dispatch / synchronization logic. This is a much better solution than relying on the old `LcmDrivenLoop` interface.
(But for the time being, I believe you can still use your custom `LcmDrivenLoop` derivative.)

That being said, I'd prefer that we close this out in lieu of Jeremys' proposals.

@mposa @jwnimmer-tri Would that work for y'all?

jwnimmer-tri, 2019-04-01:
I am OK with anything you guys want to do about this issue.  Before #10981 is done, I intend to have it show the thread-less way of using LcmDriverLoop.

mposa, 2019-04-02:
Seems good to me--though I wasn't able to look at #10981 or the recently merged PRs and understand how things were changing.

EricCousineau-TRI, 2019-04-02:
I think this comment would explain the most relevant portion:
https://github.com/RobotLocomotion/drake/pull/10981#issuecomment-475099450

Effectively, it will make getting the next time step much more explicit (and clearer) without the need for magic / needless thread synchronization.
`LcmDrivenLoop` may disappear, and in its place will be a simple recipe in documentation in how to replicate that functionality in a much more explicit and less magical way.

jwnimmer-tri, 2019-04-02:
I've been looking inside Drake to find a current use of LcmDrivenLoop that is not broken.  There are only three total right now -- one valkyrie, one iiwa, and one jaco.  The valkyrie one requires Guruobi so I can't run it easily.  The jaco one is missing pieces (what sends the plan message?).  The iiwa one maybe looks salvageable, but I have to figure out the move_iiwa_ee tool first.

If anyone can point me to some other code that uses LcmDrivenLoop in a useful way, e.g., a demo that still works at all, that might serve as a better starting point for me to illustrate the prototype for the future.

mposa, 2019-04-02:
It's far from a lightweight example, but we have a simulation that uses a dispatcher to emulate our communication with the actual Cassie robot. The simulation can be run via the procman script here: https://github.com/DAIRLab/dairlib/blob/master/examples/Cassie/cassie_test.pmd with the commands given by the script `run-drake-pd-control`.

One specific file that uses LcmDrivenLoop is here: https://github.com/DAIRLab/dairlib/blob/master/examples/Cassie/dispatcher_robot_in.cc

jwnimmer-tri, 2019-04-02:
Thanks, I'll take a look!  I still have to add one or two more APIs to Drake before it would be plausibly ready out as a prototype, but hopefully next week I'll have some progress.

PR #11281 (2019-04-22): systems/lcm: Deprecate lcm_driven_loop
Closes #5936.  Closes #10714.

---

**If you are a Drake user and would like to adjust your code to the new APIs, please read on for details.**

Drake has deprecated (and will eventually remove) the LcmDrivenLoop class, which organized the execution and message flow of an LCM-triggered control loop.

LcmDrivenLoop presents challenging problems with threading, mutexes, and condition variables because it assumes the LCM service in handled by a separate thread (no longer true as of #11166), and it is insufficiently general in the first place (#10714).

For a [... cut]
closed 2019-04-22 by jwnimmer-tri