RobotLocomotion/drake #10725: Stores actuator effort limits in MultibodyPlant

hongkai-dai, 2019-02-20:
In both URDF and SDF, it has a tag `effort` for the joint, which stores the maximal effort for the joint actuator. This information is not stored in MultibodyPlant. There is actually a TODO to store this maximal effort https://github.com/RobotLocomotion/drake/blob/ffaa0136bed506027d10487b414fa9ba113a8e55/multibody/parsing/detail_sdf_parser.cc#L203-L207

hongkai-dai, 2019-07-29:
This issue is brought up again by @huihuaTRI . Step 0: we should add a getter for the joint torque limits.

huihuaTRI, 2019-07-30:
Here is the high-level [proposal](https://docs.google.com/document/d/1cf3CVjPO6Xsw8aq3JWqDXUceFbYyZmvYdZMcxPJ_tkg/edit) for this task.

huihuaTRI, 2019-07-30:
I have put a quick implementation. But it seems to be a little bit "invasive", by which I mean the user interface has to be changed. I am not sure whether this is the right approach.
Here is the [branch](https://github.com/huihuaTRI/drake/tree/20190729_add_actuator_limit). It would be nice if @amcastro-tri or @sherm1 can take a quick look. 
I have not fixed all the unit tests yet.

huihuaTRI, 2019-07-30:
The [branch](https://github.com/huihuaTRI/drake/tree/20190729_add_actuator_limit) is updated. It is downstream compatible. Please let me know whether this will work.

sherm1, 2019-07-31:
@huihuaTRI I have some questions. It would be easier to discuss this over a `[WIP]` PR -- I looked at the branch but it would be much easier to comment in Reviewable.

huihuaTRI, 2019-12-20:
Closing this issue since the PR has been landed for a while.

closed 2019-12-20 by huihuaTRI