RobotLocomotion/drake #10728: pydrake unit tests for deprecation are too boilerplatey

jwnimmer-tri, 2019-02-20:
We currently have many stanzas that look like this:
```python
with warnings.catch_warnings(record=True) as w:
    warnings.simplefilter('always', DrakeDeprecationWarning)
    // stuff
    self.assertEqual(len(w), 2)
```

That's a lot of boilerplate.  We should condense it somehow, maybe:
```python
with ExpectDrakeDeprecation(2):
    // stuff
```

Noted in #10681 code review.

EricCousineau-TRI, 2019-02-22:
My request is that this expose the warnings `w` still, and not try to encode too much about what's to be expected, e.g.
```
with catch_drake_warnings(record=True) as w:
   // stuff
  self.assertEqual(len(w), 2)
```

Could be done relatively easily with a `contextmanager`:
https://docs.python.org/3/library/contextlib.html#contextlib.contextmanager

jwnimmer-tri, 2019-02-25:
That turns 3 lines of boilerplate into 2.  I suspect @ggould-tri would still object (as would I).

What about this?
```python
with catch_drake_warnings(action='always', expected_count=2) as w:
    pass  # stuff
```

... where the kwargs to the context are optional (defaulting to 'always', None), and so you can still do `self.assertEqual(len(w), 2)` or whatever to cover irregular conditions might occur?

EricCousineau-TRI, 2019-02-25:
Sounds good to me!

PR #10765 (2019-02-26): pydrake: De-boilerplate deprecation unit tests
Closes #10728.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10765)
closed 2019-02-26 by jwnimmer-tri