RobotLocomotion/drake #10734: MBP IK minimum distance constraint gradients incorrect for box-sphere collision

gizatt, 2019-02-20:
(Opening for tracking -- Hongkai and I have been going back and forth on this for a bit.)

In [this example](https://github.com/RobotLocomotion/drake/files/2886840/mbp_ik_collision_bug.zip), I add a fixed 10x10x10 box as the ground at translation [0, 0, -5], and a free unit-radius sphere at translation [0, 0, 0] with an off-axis rotation. In this configuration, the SignedDistancePair query seems to return something reasonable (the point pair [0, 0, 0], [0, 0, -1]).

However, I create a minimum distance constraint using the MBP IK interface, and query its gradient via some autodiff utils, and it points in a weird direction. Results of running the script on my machine are below.

```
Results of constraint evaluation: 
score:  [100.5]
qnow:  [0.07161149 0.50128041 0.5728919  0.64450339 0.         0.
 0.        ]
grad:  [  0.           0.           0.           0.          72.82051282
  66.66666667 -15.8974359 ]
*****
Results of point pair query: 
Distance -1.000000, ([0. 0. 5.] in frame 1)<->([-0.56410256 -0.81025641  0.15897436] in frame 2
Or in world frame: ([[0. 0. 0.]])<->([[-5.55111512e-17 -2.77555756e-17 -1.00000000e+00]])
```

@hongkai-dai

hongkai-dai, 2019-02-20:
Found the bug in the code, 
https://github.com/RobotLocomotion/drake/blob/1d7f9eaf9d4daf6c10368de0f60bc8e976a58b9e/multibody/inverse_kinematics/minimum_distance_constraint.cc#L113-L115
when we compute the Jacobian of point Ca in frame A, measured in frame B, this Jacobian should be expressed in the *world* frame. I incorrectly expressed it in the body frame. I will send a PR to fix this. Thanks @gizatt for creating the test!

gizatt, 2019-02-20:
Thanks for the super-quick turnaround, Hongkai!

hongkai-dai, 2019-02-21:
@gizatt could you run your test on top of #10735, to make sure the problem is gone? Thanks!

gizatt, 2019-02-21:
Can confirm that resolved it! New output is

```
Results of constraint evaluation:
score:  [100.5]
qnow:  [0.07161149 0.50128041 0.5728919  0.64450339 0.         0.
 0.        ]
grad:  [ 0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
 -5.55111512e-15 -2.77555756e-15 -1.00000000e+02]
*****
Results of point pair query:
Distance -1.000000, ([0. 0. 5.] in frame 1)<->([-0.56410256 -0.81025641  0.15897436] in frame 2
Or in world frame: ([[0. 0. 0.]])<->([[-5.55111512e-17 -2.77555756e-17 -1.00000000e+00]])
```

PR #10735 (2019-02-21): Fix the bug on the Jacobian expressed frame in MinimumDistanceConstraint
Resolves #10734

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10735)
closed 2019-02-21 by hongkai-dai