RobotLocomotion/drake #10739: maliput::utility::GenerateObj() is fragile with respect to lane width

maddog-tri, 2019-02-21:
`maliput::utility::GenerateObj()` hangs, or gives the appearance of hanging, when working on a `RoadGeometry` with a lane with a very narrow width.

This might be simply due to a defect in the `PickGridUnit()` method, which has no lower bound for the value it returns.  If the grid unit is very small compared to the overall extent of the lane, `GenerateObj()` will produce (or try to produce) a ridiculous number of quads to cover the surface.

Some YAML that will exercise this via `yaml_to_obj`:
```
# -*- yaml -*-
---
# distances are meters; angles are degrees.
maliput_multilane_builder:
  id: "ring"
  lane_width: 0.05
  elevation_bounds: [0, 5]
  scale_length: 1.0
  linear_tolerance: 0.1
  angular_tolerance: 0.5
  computation_policy: prefer-accuracy
  right_shoulder: 0
  left_shoulder: 0
  points:
    start:
      xypoint: [50, 0, 90]  # x,y, heading
      zpoint: [0, 0, -30, 0]  # z, z_dot, theta (superelevation), theta_dot
  connections:
    0:
      lanes: [1, 0, 0]
      start: ["ref", "points.start.forward"]
      arc: [50, 360]
      explicit_end: ["ref", "points.start.forward"]
```

jwnimmer-tri, 2019-02-26:
All issues must have an owner.  I've set @maddog-tri for now, but you may reassign.

liangfok, 2019-06-08:
Moved to downstream repo.

closed 2019-06-08 by liangfok