RobotLocomotion/drake #10742: Upgrade Eigen::Matrix3 to Drake RotationMatrix class.

mitiguy, 2019-02-21:
In many locations in Drake's C++ source code, an Eigen Matrix3 is currently being used to store a 3x3 rotation matrix (orientation of two frames).

This has has the following defects:

1. The name of the class (Matrix3) is not specific to its task.  A Matrix3 has a multitude of purposes completely unrelated to a RotationMatrix.  For example, adding two Matrix3 is sensible whereas adding two rotation matrices is not. 
2.  The name Matrix3 does not immediately communicate its purpose or limit its functionality.
2. The Matrix3 class does not require a valid 3x3 rotation matrix (Matrix3 is not necessarily orthonormal).
3. The Matrix3 class is missing much functionality and "sugar" that is specific to rotation matrices, e.g., conversion to roll-pitch-yaw angles, or Quaternions, or ...

Matrix3 is being replaced by RotationMatrix, which has the following advantages:

1. In debug versions, RigidTransform requires a valid 3x3 rotation matrix or it throws an assertion.
2. RotationMatrix has significantly more functionality than Matrix3.
3. Implementing the RotationMatrix class has caught  bugs in Drake that were previously undetected.
4. RotationMatrix can (or has been) "pythonized".

jwnimmer-tri, 2021-11-12:
@mitiguy What's your plan here?  Maybe create a checklist of locations that need fixing?  I just spent a few minutes scanning through the code, and I couldn't find any.

jwnimmer-tri, 2022-04-25:
Given the lack of response, I'll presume that all of the important locations have already been fixed.  Closing now.

closed 2022-04-25 by jwnimmer-tri

mitiguy, 2022-04-25:
@jwnimmer-tri    Yes -- thank you for closing this.