RobotLocomotion/drake #10744: Add tests for JacobianWrtVariable::kV in multibody_plant_jacobians_test.cc

mitiguy, 2019-02-22:
Currently, the test KukaIiwaModelTests in file multibody_plant_jacobians_test.cc only tests for JacobianWrtVariables::kQDot.  This test was overlooked with more than one developer in previous PRs.

One possible solution involves adding a method that leverages MapVelocityToQDot() in file multibody_tree.h to form the "N" matrix that relates qDot to v and use this method inAdd a method in conjunction with the analytical calculations that are already performed for kQDot.

mitiguy, 2020-07-20:
In file multibody_plant_jacobians_test.cc.
In test functions such as:  TEST_F(KukaIiwaModelTests, CalcJacobianTranslationalVelocityNonUnitQuaternion)

Add methods to test
CalcJacobianSpatialVelocity(*context, JacobianWrtVariable::kV, frame_E,  p_EoEp, frame_W, frame_W, &Jv_V_WEp_W);

CalcJacobianTranslationalVelocity(*context, JacobianWrtVariable::kV, frame_E,  p_EoEp_E, frame_W, frame_W, &Jq_v_WEp_W);

jwnimmer-tri, 2024-03-20:
If we haven't missed the tests in the past 4 years, chances are we don't care.  Closing as not planned.

closed 2024-03-20 by jwnimmer-tri