RobotLocomotion/drake #10745: py systems: Enable / show example of doing convertible VectorSystem

EricCousineau-TRI, 2019-02-22:
From @RussTedrake's pain points:

> RuntimeError: The object named [] of type drake::pydrake::(anonymous)::Impl<double>::PyVectorSystem does not support ToAutoDiffXd.

There are examples of doing scalar convertible types here:
https://github.com/RobotLocomotion/drake/tree/ffaa013/bindings/pydrake/systems/test/scalar_conversion_test.py#L12

However, I should inspect how hard it is to allow intermediate C++ base classes (e.g. `VectorSystem`) to do scalar conversion through this shindig.

RussTedrake, 2019-02-24:
Thanks!  My "pain point" request has two parts.  (1) just being able to convert subclasses of VectorSystem... which I think is just a matter of declaring the ScalarConverter versions of the constructor, etc(?), and (2) getting rid of the boilerplate around the python system conversion.  

To be concrete, I was hoping to write this:
```
from pydrake.all import LinearQuadraticRegulator, VectorSystem

class Quadrotor2D(VectorSystem):
    def __init__(self):
        # two inputs (thrust), six outputs (full state)
        VectorSystem.__init__(self, 2, 6)
        # three positions, three velocities
        self._DeclareContinuousState(3, 3, 0)

        self.L = 0.25;    # length of rotor arm
        self.m = 0.486;   # mass of quadrotor
        self.I = 0.00383; # moment of inertia
        self.g = 9.81;    # gravity
        
    def _DoCalcVectorOutput(self, context, u, x, y):
        y[:] = x
        
    def _DoCalcVectorTimeDerivatives(self, context, u, x, xdot):
        xdot[0:3] = x[3:6]
        xdot[4] = -np.sin(x[2])/self.m*(u[0] + u[1])
        xdot[5] = -self.g + np.cos(x[2])/self.m*(u[0] + u[1])
        xdot[6] = self.L/self.I*(-u[0] + u[1])
        
plant = Quadrotor2D()

context = plant.CreateDefaultContext()
context.SetContinuousState(np.zeros([6, 1]))
context.FixInputPort(0, plant.m*plant.g/2.*np.array([1, 1]))

Q = np.diag([10, 10, 10, 1, 1, (plant.L/2./np.pi)])  
R = np.array([[0.1, 0.05], [0.05, 0.1]])  

pi = LinearQuadraticRegulator(plant, context, Q, R)
```

but had to (i think) instead write this
```
from pydrake.all import BasicVector_, LeafSystem_, LinearQuadraticRegulator, TemplateSystem 

@TemplateSystem.define("Quadrotor2D_")
def Quadrotor2D_(T):

    class Impl(LeafSystem_[T]):
        def _construct(self, converter=None):
            LeafSystem_[T].__init__(self, converter)
            # two inputs (thrust)
            self._DeclareVectorInputPort("u", BasicVector_[T](2))
            # six outputs (full state)
            self._DeclareVectorOutputPort("x", BasicVector_[T](6), self._CopyStateOut)
            # three positions, three velocities
            self._DeclareContinuousState(3, 3, 0)

            self.L = 0.25;    # length of rotor arm
            self.m = 0.486;   # mass of quadrotor
            self.I = 0.00383; # moment of inertia
            self.g = 9.81;    # gravity

        def _construct_copy(self, other, converter=None):
            Impl._construct(self, converter=converter)
            
        def _CopyStateOut(self, context, output):
            x = context.get_continuous_state_vector().get_value()
            y = output.get_mutable_value()
            y[:] = x

        def _DoCalcTimeDerivatives(self, context, derivatives):
            x = context.get_continuous_state_vector().CopyToVector()
            u = self.EvalVectorInput(context, 0).CopyToVector()
            q = x[:3]
            qdot = x[3:]
            qddot = np.array([-np.sin(q[2])/self.m*(u[0] + u[1]), 
                              -self.g + np.cos(x[2])/self.m*(u[0] + u[1]),
                              self.L/self.I*(-u[0] + u[1])])
            derivatives.get_mutable_vector().SetFromVector(np.concatenate((qdot, qddot)))

    return Impl
  
Quadrotor2D = Quadrotor2D_[None]  # Default instantiation            
            
plant = Quadrotor2D()

context = plant.CreateDefaultContext()
context.SetContinuousState(np.zeros([6, 1]))
context.FixInputPort(0, plant.m*plant.g/2.*np.array([1, 1]))

Q = np.diag([10, 10, 10, 1, 1, (plant.L/2./np.pi)])  
R = np.array([[0.1, 0.05], [0.05, 0.1]])  

pi = LinearQuadraticRegulator(plant, context, Q, R)
```

BTW -- love that the second one worked!

EricCousineau-TRI, 2019-02-25:
I would like this issue to focus on the first pain point, will create a separate one for the second. Thanks!

EricCousineau-TRI, 2019-08-19:
User ran into this pain point again on StackOverflow: https://stackoverflow.com/questions/57512465/class-implementation-in-python

Upon revisiting this, though, it seems like it may be more complicated: `VectorSystem` overloads depend on reference semantics between C++ Eigen matrices and Python NumPy arrays, which is still not supported (https://github.com/RobotLocomotion/drake/issues/8116).

That issue is a still be nebulous due to the NumPy version requirements + potential for segfaults.

Proposed solutions:
* First try to make the error message encountered a little more tractable for users.
* Briefly see what's involved in the dtype stuff - knowing that it may open a can of worms.
* Try out @jwnimmer-tri's suggestion in making a pure Python sugar class, and ditch overloading of the C++ `VectorSystem` in Python.
  * Caveat here is any analysis that is constrained to `VectorSystem`; I should see what methods those are.

jwnimmer-tri, 2024-03-16:
Per https://github.com/RobotLocomotion/drake/issues/20989#issuecomment-2002129534, we should disprefer VectorSystem in the first place, so there's no point in gold-plating its Python bindings.

closed 2024-03-16 by jwnimmer-tri