RobotLocomotion/drake #10747: return the normal direction when query the signed distance

hongkai-dai, 2019-02-22:
Currently when we query the SignedDistancePair, it only returns the geometry ID and the contact points on the body. It doesn't return the normals (The normal can be understood as the gradient of the signed distance field). When the two geometries are not touching, we can compute the contact normal as `(p1 - p2).normalized()`, where `p1, p2` are the contact points on the two bodies. But when the two objects are touching (which happens a lot if we use rigid contact), then we cannot normalize the zero vector `p1 - p2`.

On the other hand, when the two objects are touching, the surface normal could serve as a meaningful normal.

DamrongGuoy, 2019-02-22:
Could we prioritize the kinds of pairs of geometries?  If it's sphere v.s. {sphere, box, cylinder}, it could be done quickly.  If it's box v.s. box, it might take a bit longer.  If it's all v.s. all, I'll need to check the code to estimate.

sherm1, 2019-02-23:
Per f2f w @DamrongGuoy and @hongkai-dai: the intent of this is just to expose the normals that Damrong is _already_ calculating in the point-based distance queries, when executing the pairwise method that expands the points into spheres. This will require either a new API, or (better, if we can get away with it) just enhance the returned struct to include the normal, with TODOs about any cases where we don't have one.

DamrongGuoy, 2019-03-07:
status: We will fix this issue in two PRs.  The first PR #10823 is to define API and unblock @hongkai-dai, and it is under code review.  The second PR will start after another in-flight PR #10813 landed to avoid code conflict.

sherm1, 2019-03-27:
status: #10823 merged to unblock @hongkai-dai

DamrongGuoy, 2019-04-29:
This issue will be solved by PR #11272, which is paused right now and will resume when @SeanCurtis-TRI  finish refactoring proximity_engine.cc.

tri-ltyyu, 2019-05-06:
@SeanCurtis-TRI, is proximity_engine.cc refactoring part of the getting rendering into SG work? Or is it track elsewhere? Just want to see if @SeanCurtis-TRI is aware that there's a dependency here.

SeanCurtis-TRI, 2019-05-06:
I'm aware of the dependency; the blocking PR is currently under review (and it's built into my OKRs...so all accounted for.)

DamrongGuoy, 2019-05-06:
I see @SeanCurtis-TRI have #11379 almost done with the code review.  I'll resume my work now.

PR #11435 (2019-05-13): [geometry] Return normal for touching sphere-{sphere, box, cylinder}.
Fix #10747.  Most of the work was done by @SeanCurtis-TRI as a part of #11379 already.  I just added `is_nhat_BA_W_unique` to `SignedDistancePair` and set up unit tests.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11435)
closed 2019-05-13 by DamrongGuoy