RobotLocomotion/drake #10761: Relocate //systems/analysis:lyapunov?

jwnimmer-tri, 2019-02-26:
To avoid unwanted accidental dependencies like #10705, I wonder if `//systems/analysis:lyapunov` should be relocated?  Would `//systems/optimization:lyapunov` be a more meaningful home?  As it stands now, it's the only thing in `analysis` that brings in any solvers, and the solvers package is a major subsystem, where we should be careful about spuriously adding into build targets.

WDYT @RussTedrake @avalenzu?  I think the main question is whether the new home would be more sensible or not from a discoverability / expectations perspective.  If it's best placed where it currently lives, then I can look at the dependency hazard mitigations separately, but if moving it makes sense that's the easy answer.

avalenzu, 2019-02-26:
Hmm. I'm not sure I agree with that move. Yes, `SampleBasedLyapunovAnalysis()` solves an optimization problem, but it *is* a way of analyzing a system. I don't think that we should let the simulator push all usage of `solvers` out of `analysis`.

RussTedrake, 2019-02-26:
I agree with andres.  I’d be more inclined to move simulator out of analysis, since it feels more out of place than the optimization approaches (for which there will be many more).

jwnimmer-tri, 2019-03-01:
OK, I'll definitely keep `:lyapunov` where it is.  Next time I come back to this question, I'll try to figure out the most sensible way to help control / reduce dependency waste.

closed 2019-03-01 by jwnimmer-tri