RobotLocomotion/drake #10769: pybind: Add `type_caster` for EigenPtr

EricCousineau-TRI, 2019-02-26:
From Slack convo: https://drakedevelopers.slack.com/archives/C3YB3EV5W/p1551220325006300

\cc @jwnimmer-tri

jwnimmer-tri, 2019-02-27:
For the record, the easy work-around for now is to bind as `Eigen::Ref<...> arg`, as long as nullptr is not relevant (and then pass `&arg` to the C++ function).

jwnimmer-tri, 2022-02-14:
We have seen some increasing use of `EigenPtr` vs our bindings, so it would be nice to have this type_caster to remove some boilerplate.

For example, here is one recent use of the work-around:

https://github.com/RobotLocomotion/drake/blob/2f75971b66ca59dc2c1dee4acd78952474936a79/bindings/pydrake/systems/primitives_py.cc#L351-L359

BetsyMcPhail, 2022-02-17:
@jwnimmer-tri would a good demonstration of the failure for this be to replace a Eigen::Ref workaround with EigenPtr in the bindings and see what fails? Do you have a specific example that would be good to try?

PR #16625 (2022-02-25): Implement a pybind11 type_caster for drake::EigenPtr
Fixes: #10769

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16625)
closed 2022-02-25 by jwnimmer-tri