RobotLocomotion/drake #10775: Parse urdf/sdf into SceneGraph (only)

RussTedrake, 2019-02-27:
I'm finding myself with several use cases where I would like to add content from a urdf/sdf directly into a SceneGraph.  Right now, I'm parsing into an MBP, then throwing that MBP away.  

There is an existing example from @jwnimmer-tri (with a TODO) here:
https://github.com/RobotLocomotion/drake/blob/master/automotive/automotive_simulator.cc#L194

The example I'm working on right now (PR coming shortly) is 
https://github.com/RobotLocomotion/drake/blob/master/examples/quadrotor/run_quadrotor_dynamics.cc#L52
I'm teaching QuadrotorPlant to populate SceneGraph (by parsing into a temporary MultibodyPlant), but don't want to have to do it *again* just to add the warehouse geometry.

sherm1, 2019-02-27:
I believe everyone is tired of hearing me say this, but I think it is a mistake to entangle file parsing with the objects that need the parsed information. If we parse into an intermediate format rather than directly into MBP and SG, then those objects can be asked to extract what they want from the intermediate format.

jwnimmer-tri, 2019-02-27:
If you whiteboard out how that looks in this use case, I don't think it actually ends up any better.

SeanCurtis-TRI, 2019-02-27:
@EricCousineau-TRI and I have had preliminary discussions on this very topic. The seed to our initial meeting is [here](https://docs.google.com/document/d/1zs5Byy_BBB8NqtTYhsQzva9Upk16zVxlKei9g8U6UJ0/edit#heading=h.zeprpi6ddq8r).

I still need to write up some pseudocode outlining the API and discuss this particular type of use case. This might finally be the stick to get me moving on this.

EricCousineau-TRI, 2019-04-08:
Sorry for the late response here; we have discussed on intermediate formats, but I do fear that usability may suffer as @jwnimmer-tri mentioned.

But yeah, my pref. is that we do some more whiteboarding before pulling the trigger.

jwnimmer-tri, 2022-05-03:
For reference, the existing code is here:

https://github.com/RobotLocomotion/drake/blob/0f7965c851470c7d277548e495e277b7aa52859e/examples/quadrotor/quadrotor_geometry.cc#L41-L53

closed 2024-03-20 by jwnimmer-tri