RobotLocomotion/drake #10777: MeshcatVisualizer doesn't handle mesh scale

RussTedrake, 2019-02-27:
leading to a comically large quadrotor in meshcat:
https://github.com/RussTedrake/underactuated/blob/master/src/quadrotor/lqr.py
https://github.com/RobotLocomotion/drake/blob/master/examples/quadrotor/quadrotor.urdf#L17

the information is available to the MeshcatVisualizer class in the lcm message
https://github.com/RobotLocomotion/drake/blob/master/geometry/geometry_visualization.cc#L106

but I don't *think* meshcat-python lets me send it through, yet:
https://github.com/rdeits/meshcat-python/blob/master/src/meshcat/geometry.py#L314

my simple attempt (adding to meshcat_visualizer line 211):
```
                    element_local_tf[3,:3] = geom.float_data[:3]
```
didn't seem to work.

@gizatt -- might you take a quick look to see if it's an easy fix?

gizatt, 2019-02-27:
What you tried was really close to a solution -- didn't work for me either. ```element_local_tf[:3, :3] = element_local_tf[:3, :3].dot(np.diag(geom.float_data[:3]))``` did, though.

It doesn't look like `meshcat-python` supports the real three.js scale interface yet (mentioned [here](https://github.com/mrdoob/three.js/issues/991)), though I think it'd be a trivial addition. I can open a PR with this slightly hacky fix to fix this the fastest, and I'll leave a TODO to look out for improved support on the meshcat-python side?

RussTedrake, 2019-02-27:
sounds perfect.  thanks Greg.

RussTedrake, 2020-11-02:
This was resolved by @gizatt long ago.

closed 2020-11-02 by RussTedrake