RobotLocomotion/drake #10780: Visualize multiple postures for the same MBP in the drake_visualizer

hongkai-dai, 2019-02-27:
Currently we can only visualize a single posture of the robot. If we want to visualize multiple postures of the same robot simultaneously (like I want to visualize the way points of a planned motion), there is not a clean solution to that.

Per f2f discussion with @SeanCurtis-TRI , we need to define a new LCM message to wrap the multiple postures, and drake-visualize need to parse that LCM message.

jwnimmer-tri, 2019-02-27:
By "drake-visuaslizer needs to parse that message", I hope you mean that we write a https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/drake_visualizer/plugin/ that does this?

I suspect if you talk to @avalenzu or @calderpg-tri maybe we already have a solution for this?  I think we are moving to use rvis more and more anyway.  Not sure how much we really should sink cost into drake-visualizer enhancements.

edrumwri, 2019-02-28:
On a related note, we need to visualize multiple things from the
hydrostatic contact model. I'd hate to invest time into
LCM/drake-visualizer if we're moving toward rviz. How long until Drake has
ROS support?

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

SeanCurtis-TRI, 2019-02-28:
@edrumwri Please create a new issue outlining the features you want and then please append it to the list in #10786. I'd like to keep different features in different issues so the conversations don't get muddied.

calderpg-tri, 2019-02-28:
Visualizing multiple postures is already supported in the RViz interface in anzu, and my guess is that the things @edrumwri wants to visualize for the hydrostatic model are easy to visualize via [RViz markers](http://wiki.ros.org/rviz/DisplayTypes/Marker).

Anzu has some ROS support already; Drake ROS support is waiting on testing of ROS2. ROS2 has much better Mac support than ROS1 so if we can avoid ROS1 support in Drake we will.

EricCousineau-TRI, 2020-10-27:
Relates #12702

jwnimmer-tri, 2021-12-14:
@joemasterjohn @hongkai-dai I propose closing this as a"WONTFIX" per #16215. What do you think?

hongkai-dai, 2021-12-14:
Agreed, we should close this issue

closed 2021-12-14 by hongkai-dai