RobotLocomotion/drake #10786: Advanced visualization features

SeanCurtis-TRI, 2019-02-27:
This is an epic issue that tracks various "advanced" visualization requests. In this case, "advanced" means anything that is not a simple, literal visualization of the instantaneous state of the registered geometries. The preferred use of this issue is to create a *separate* issue with the detailed feature request (which will also serve as the platform for discussing the issue) and only link to it here with some form of summary that will allow other feature requesters to reasonably assess if their request has already been added. 

  - [ ] #10780 Visualize multiple postures for the same MBP in the drake_visualizer 
    - Given a robot with corresponding registered illustration geometry, we want to be able to visualize multiple instances of that geometry in several poses simultaneously.

RussTedrake, 2021-07-06:
Possibly related: #13038

jwnimmer-tri, 2023-07-30:
It does not seem to me like this issue is actionable.  It's only purpose in life is as a table of contents for other issues?  It seems like there must be better ways to do that.  How about we close this one?

closed 2024-03-20 by jwnimmer-tri