RobotLocomotion/drake #10787: Enhance ports viz with System::GetGraphvizString()

amcastro-tri, 2019-02-27:
I was trying to debug the manipulation station today using `System::GetGraphvizString()`. In particular, I was trying to figure out if any of the exported input ports were left unconnected in one of the unit tests. 

This is what the input ports to `ManipulationStation` look like in the generated `dot` file:
[image]

I'd like to notice two missing pieces of information that'd be super helpful to debug these cases:
- Exported input ports (output as well actually) are not named. They actually were given nice names and it'd be great showing those.
- It'd be nice having some color coding to find out which ports are not connected. Say, green for connected, red for disconnected and orange for fixed.
- (maybe) color code arrows, given that when they overlap it's difficult to tell where the are coming from.

Note: For fixed ports we'd need `GetGraphvizString()` to take a `Context`. @jwnimmer-tri pointed out that we could even show port values if we pass a context. Though I'm personally happy with names and colors for now.

avalenzu, 2019-02-28:
FWIW, if you call `diagram.GetGraphvizString(1)`for the diagram containing the `ManipulationStation` you can see the port names and it's obvious which ports are connected/disconnected. You're right though, that you wouldn't be able to tell if a port were fixed in any particular Context.

That being said, I agree that showing the port names at the System level would be good too.

jwnimmer-tri, 2023-06-14:
Another reminder for next time we're in this area -- in some cases there are _way too many_ ports in the figure (e.g., if you have lots of model instances on a MbP).  We might also consider adding an option to GetGraphvizString to _omit_ ports that are not connected from the figure, to help reduce clutter.

jwnimmer-tri, 2024-03-16:
Fixed in #20204, with a completely new look.  Please open a new issue if there are still any difficulties reading the graphviz information.

closed 2024-03-16 by jwnimmer-tri