RobotLocomotion/drake #10788: fast_math_test coverage test times out

liangfok, 2019-02-27:
Coverage unit test `//tools/install/libdrake:fast_math_test` times out:

* https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-bionic-clang-bazel-nightly-everything-coverage/107/
* https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-bionic-clang-bazel-nightly-coverage/104/
* https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-bionic-clang-bazel-nightly-coverage/103/

Because of this, it was removed from the coverage tests. This tracks the process of root-causing and fixing the problem.

jwnimmer-tri, 2019-03-01:
@EricCousineau-TRI worth fixing?  I'd vote no -- we don't really need coverage reports for this (tiny) test case.

EricCousineau-TRI, 2019-03-01:
I'd say not for now, as it is really small. Can re-open if it becomes a future.

Thanks @liangfok for the kcov tag fix!

closed 2019-03-01 by EricCousineau-TRI